zephyrproject-rtos/zephyr #20494: [Coverity CID :205617]Memory - corruptions in /drivers/kscan/kscan_mchp_xec.c

aasthagr, 2019-11-08:
Static code scan issues seen in File: /drivers/kscan/kscan_mchp_xec.c
 Category: Memory - corruptions
 Function: check_key_events
 Component: Drivers
 CID: 205617
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aescolar, 2019-11-11:
0d85074d9a9
```
154static bool check_key_events(void *dev)
155{
156        u8_t matrix_new_state[MAX_MATRIX_KEY_COLS] = {0U};
157        bool key_pressed = false;
158        u32_t cycles_now  = k_cycle_get_32();
159
   	1. Condition ++kbd_data.scan_cycles_idx > 30U, taking false branch.
   	2. cond_at_most: Checking ++kbd_data.scan_cycles_idx > 30U implies that kbd_data.scan_cycles_idx may be up to 30 on the false branch.
160        if (++kbd_data.scan_cycles_idx > SCAN_OCURRENCES)
161                kbd_data.scan_cycles_idx = 0U;
162
   	
CID 205617 (#1 of 1): Out-of-bounds write (OVERRUN)
3. overrun-local: Overrunning array kbd_data.scan_clk_cycle of 30 bytes at byte offset 30 using index kbd_data.scan_cycles_idx (which evaluates to 30).
163        kbd_data.scan_clk_cycle[kbd_data.scan_cycles_idx] = cycles_now;
164
165        /* Scan the matrix */
166        key_pressed = read_keyboard_matrix(matrix_new_state);
167
168        /* Abort if ghosting is detected */
169        if (is_matrix_ghosting(matrix_new_state)) {
170                return false;
171        }
172
```

nashif, 2019-11-27:
@franciscomunoz  can you please address this one?

dleach02, 2019-12-02:
```
160        if (++kbd_data.scan_cycles_idx > SCAN_OCURRENCES)
161                kbd_data.scan_cycles_idx = 0U;
```

this should be >=

PR #21108 (2019-12-02): drivers: kscan_mchp_xec: fix memory corruption
Fixes memory corruption (Out-of-bounds write ) case discovered by coverity.

Fixes #20494
Coverity CID: 205617
commit d0aa6fc2 (2019-12-02): drivers: kscan_mchp_xec: fix memory corruption

Fixes memory corruption (Out-of-bounds write ) case discovered by
coverity.

Fixes #20494
Coverity CID: 205617
closed 2019-12-02 by dleach02