zephyrproject-rtos/zephyr #20496: [Coverity CID :205630]Memory - illegal accesses in /drivers/pwm/pwm_mchp_xec.c

aasthagr, 2019-11-08:
Static code scan issues seen in File: /drivers/pwm/pwm_mchp_xec.c
 Category: Memory - illegal accesses
 Function: xec_compare_params
 Component: Drivers
 CID: 205630
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aescolar, 2019-11-11:
1ee9e1046a6
```
195static struct xec_params *xec_compare_params(u32_t target_freq,
196                                             struct xec_params *hc_params,
197                                             struct xec_params *lc_params)
198{
199        struct xec_params *params;
200        u32_t freq_h = 0;
201        u32_t freq_l = 0;
202
   	1. Condition hc_params->div < 255, taking true branch.
   	2. cond_at_most: Checking hc_params->div < 255 implies that hc_params->div may be up to 254 on the true branch.
203        if (hc_params->div < UINT8_MAX) {
   	
CID 205630 (#1 of 1): Out-of-bounds read (OVERRUN)
3. overrun-local: Overrunning array max_freq_high_on_div of 16 4-byte elements at element index 254 (byte offset 1019) using index hc_params->div (which evaluates to 254).
204                freq_h = xec_compute_frequency(
205                                max_freq_high_on_div[hc_params->div],
206                                hc_params->on,
207                                hc_params->off);
208        }
209
210        if (lc_params->div < UINT8_MAX) {
   	CID 205622: Out-of-bounds read (OVERRUN) [select issue]
211                freq_l = xec_compute_frequency(
212                                max_freq_low_on_div[lc_params->div],
213                                lc_params->on,
214                                lc_params->off);
215        }
216
   	CID 205638: Macro compares unsigned to 0 (NO_EFFECT) [select issue]
217        if (abs(target_freq - freq_h) < abs(target_freq - freq_l)) {
218                params = hc_params;
219        } else {
220                params = lc_params;
221        }
222
223        LOG_DBG("\tFrequency (x%u): %u", XEC_PWM_FREQ_PF, freq_h);
224        LOG_DBG("\tOn %s clock, ON %u OFF %u DIV %u",
225                params == hc_params ? "High" : "Low",
226                params->on, params->off, params->div);
227
228        return params;
229}
```

dleach02, 2019-11-22:
Not sure why the test is looking for < UINT8_MAX? Seems like the code was attempting to ensure that the access doesn't go past the end of the max_freq_high_on_div[] array so do a test against ARRAY_SIZE(max_freq_low_on_div)

commit ceb2303d (2019-11-27): pwm: pwm_mchp_xec: fix potential out of bound access

Fix an issue discovered by Coverity where there is a potential
out of bound access on the divisor arrays.

Fixes #20495
Fixes #20496
closed 2019-11-27 by nashif