zephyrproject-rtos/zephyr #20499: [Coverity CID :205625]Error handling issues in /drivers/sensor/lsm6dso/lsm6dso_shub.c

aasthagr, 2019-11-08:
Static code scan issues seen in File: /drivers/sensor/lsm6dso/lsm6dso_shub.c
 Category: Error handling issues
 Function: lsm6dso_shub_embedded_en
 Component: Drivers
 CID: 205625
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aescolar, 2019-11-11:
72ae24d8d59
```
426static inline void lsm6dso_shub_embedded_en(struct lsm6dso_data *data, bool on)
427{
   	1. Condition on, taking true branch.
428        if (on) {
   	
CID 205625 (#1 of 1): Unchecked return value (CHECKED_RETURN)
2. check_return: Calling lsm6dso_mem_bank_set without checking return value (as is done elsewhere 31 out of 32 times).
429                lsm6dso_mem_bank_set(data->ctx, LSM6DSO_SENSOR_HUB_BANK);
430        } else {
431                lsm6dso_mem_bank_set(data->ctx, LSM6DSO_USER_BANK);
432        }
433
434        k_busy_wait(150);
435}
```

avisconti, 2019-11-11:
Is it really necessary to check return value of lsm6dso_mem_bank_set() when lsm6dso_shub_embedded_en() is not returning any value?

avisconti, 2019-11-11:
> Is it really necessary to check return value of lsm6dso_mem_bank_set() when lsm6dso_shub_embedded_en() is not returning any value?

OK, I will cast it with (void) as it seems a suggested solution.

MaureenHelm, 2019-11-14:
Fixed by #20615

closed 2019-11-14 by MaureenHelm