zephyrproject-rtos/zephyr #20500: [Coverity CID :205629]Control flow issues in /drivers/timer/cc13x2_cc26x2_rtc_timer.c

aasthagr, 2019-11-08:
Static code scan issues seen in File: /drivers/timer/cc13x2_cc26x2_rtc_timer.c
 Category: Control flow issues
 Function: startDevice
 Component: Drivers
 CID: 205629
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aescolar, 2019-11-11:
```
143static void startDevice(void)
144{
145        u32_t compare;
146        u64_t period;
147        unsigned int key;
148
149        key = irq_lock();
150
151        /* reset timer */
152        AONRTCReset();
153        AONRTCEventClear(AON_RTC_CH0);
154        IntPendClear(INT_AON_RTC_COMB);
155
156        /*
157         * set the compare register to one period.
158         * For a very small period round up to interrupt upon 4th tick in
159         * compare register
160         */
   	assignment: Assigning: period = 131072ULL.
161        period = RTC_COUNTS_PER_TICK;
   	const: At condition period < 262144ULL, the value of period must be equal to 131072.
   	dead_error_condition: The condition period < 262144ULL must be true.
162        if (period < 0x40000) {
163                compare = 0x4; /* 4 * 15.5us ~= 62us */
164        } else {
165                /* else, interrupt on first period expiration */
   	
CID 205629 (#1 of 1): Logically dead code (DEADCODE)
dead_error_line: Execution cannot reach this statement: compare = period >> 16;.
166                compare = period >> 16;
167        }
168
169        /* set the compare value at the RTC */
170        AONRTCCompareValueSet(AON_RTC_CH0, compare);
171
172        /* enable compare channel 0 */
173        AONEventMcuWakeUpSet(AON_EVENT_MCU_WU0, AON_EVENT_RTC0);
174        AONRTCChannelEnable(AON_RTC_CH0);
175        AONRTCCombinedEventConfig(AON_RTC_CH0);
176
177        /* start timer */
178        AONRTCEnable();
179
180        irq_unlock(key);
181}
```
aeb8d017b5c

vanti, 2019-11-14:
@galak @aasthagr Not sure how best to deal with this one. 'period' is set to RTC_COUNTS_PER_TICK, which is dependent on the Kconfig CONFIG_SYS_CLOCK_TICKS_PER_SEC. Even though in practice most people would stick with the latter's default value, it is configurable, and depending on its value, line 166 could be run.

galak, 2019-11-19:
@vanti you can mark it as a false positive in coverity, with the statement you have here about that because of Kconfig this may vary.

vanti, 2019-11-19:
Thanks @galak. Added the info in the Triage section on Coverity.

dleach02, 2020-07-24:
closing due to false positive determination

closed 2020-07-24 by dleach02