zephyrproject-rtos/zephyr #20502: [Coverity CID :205665]Integer handling issues in /drivers/video/mt9m114.c

aasthagr, 2019-11-08:
Static code scan issues seen in File: /drivers/video/mt9m114.c
 Category: Integer handling issues
 Function: mt9m114_write_reg
 Component: Drivers
 CID: 205665
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aasthagr, 2019-11-08:
```
*** CID 205665:  Integer handling issues  (CONSTANT_EXPRESSION_RESULT)
/drivers/video/mt9m114.c: 143 in mt9m114_write_reg()
137     
138        switch (reg_size) {
139        case 2:
140                        *(u16_t *)value = sys_cpu_to_be16(*(u16_t *)value);
141                        break;
142        case 4:
     CID 205665:  Integer handling issues  (CONSTANT_EXPRESSION_RESULT)
     "*((u16_t *)value) >> 24" is 0 regardless of the values of its operands. This occurs as the bitwise first operand of "&".
143                        *(u16_t *)value = sys_cpu_to_be32(*(u16_t *)value);
144                        break;
145        case 1:
146                        break;
147        default:
148                        return -ENOTSUP;
```

aescolar, 2019-11-11:
92ee992e7cc

PR #20706 (2019-11-15): video: mt9m114: Fix integer handling issues
Fix 32-bit endian conversion.

CID: 205643

Fixes #20501 
Fixes #20502