zephyrproject-rtos/zephyr #20503: [Coverity CID :205655]Error handling issues in /drivers/watchdog/wdt_wwdg_stm32.c

aasthagr, 2019-11-08:
Static code scan issues seen in File: /drivers/watchdog/wdt_wwdg_stm32.c
 Category: Error handling issues
 Function: wwdg_stm32_get_pclk
 Component: Drivers
 CID: 205655
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aasthagr, 2019-11-08:
```
*** CID 205655:  Error handling issues  (CHECKED_RETURN)
/drivers/watchdog/wdt_wwdg_stm32.c: 69 in wwdg_stm32_get_pclk()
63          struct device *clk = device_get_binding(STM32_CLOCK_CONTROL_NAME);
64          const struct wwdg_stm32_config *cfg = WWDG_STM32_CFG(dev);
65          u32_t pclk_rate;
66     
67          __ASSERT_NO_MSG(clk);
68     
     CID 205655:  Error handling issues  (CHECKED_RETURN)
     Calling "clock_control_get_rate" without checking return value (as is done elsewhere 9 out of 10 times).
69          clock_control_get_rate(clk, (clock_control_subsys_t *) &cfg->pclken,
70                                                 &pclk_rate);
71     
72          return pclk_rate;
73     }
74   
```

FRASTM, 2019-11-12:
wwdg_stm32_get_pclk changed to test the return value of clock_control_get_rate

commit 458a8a0a (2019-11-18): driver: pwm: Error handling issues in pwm_stm32_get_cycles_per_sec

This patch tests the return code when calling clock_control_get_rate
and completes the issue  #20503 seen on watchdog
[Coverity CID :205655]

Fixes #20503
commit a0bd434f (2019-11-18): driver: spi: Error handling issues in spi_stm32_configure

This patch tests the return code when calling clock_control_get_rate
and completes the issue  #20503 seen on watchdog
[Coverity CID :205655]

Fixes #20503
commit d3ffa8d0 (2019-11-18): driver: serial: Error handling issues in uart_stm32_set_baudrate

This patch tests the return code when calling clock_control_get_rate
and completes the issue #20503 seen on watchdog
[Coverity CID :205655]

Fixes #20503
commit fc06346a (2019-11-18): driver: watchdog: Error handling issues in wdt_wwdg_stm32.c #20503

This patch tests the return code when calling clock_control_get_rate
[Coverity CID :205655]

Fixes #20503
closed 2019-11-18 by MaureenHelm