zephyrproject-rtos/zephyr #20506: [Coverity CID :205672]Null pointer dereferences in /samples/drivers/espi/src/main.c

aasthagr, 2019-11-08:
Static code scan issues seen in File: /samples/drivers/espi/src/main.c
 Category: Null pointer dereferences
 Function: main
 Component: Samples
 CID: 205672
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aasthagr, 2019-11-08:
```
*** CID 205672:  Null pointer dereferences  (FORWARD_NULL)
/samples/drivers/espi/src/main.c: 247 in main()
241                        printk("%s FOUND!\n", CONFIG_ESPI_DEV);
242        }
243     
244        printk("Hello eSPI test! %s\n", CONFIG_BOARD);
245     
246     #ifdef CONFIG_ESPI_GPIO_DEV_NEEDED
     CID 205672:  Null pointer dereferences  (FORWARD_NULL)
     Passing null pointer "gpio_dev0" to "gpio_pin_configure", which dereferences it.
247        ret = gpio_pin_configure(gpio_dev0, CONFIG_PWRGD_PIN, GPIO_DIR_IN);
248        if (ret) {
249                        printk("Unable to configure PWRGD %d\n", CONFIG_PWRGD_PIN);
250                        return;
251        }
252     
```

aescolar, 2019-11-11:
953952adbd9

closed 2019-11-15 by ioannisg