zephyrproject-rtos/zephyr #20507: [Coverity CID :205662]Memory - illegal accesses in /samples/net/sockets/dumb_http_server_mt/src/main.c

aasthagr, 2019-11-08:
Static code scan issues seen in File: /samples/net/sockets/dumb_http_server_mt/src/main.c
 Category: Memory - illegal accesses
 Function: client_conn_handler
 Component: Samples
 CID: 205662
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aasthagr, 2019-11-08:
```
*** CID 205662:  Memory - illegal accesses  (STRING_NULL)
/samples/net/sockets/dumb_http_server_mt/src/main.c: 168 in client_conn_handler()
162                        * be used in production code. This is done like this just
163                        * for this sample application to keep things simple.
164                        *
165                        * We are assuming here that the full HTTP request is received
166                        * in one TCP segment which in real life might not.
167                        */
     CID 205662:  Memory - illegal accesses  (STRING_NULL)
     Passing unterminated string "buf" to "strstr", which expects a null-terminated string. [Note: The source code implementation of the function has been overridden by a builtin model.]
168                        if (strstr(buf, "\r\n\r\n")) {
169                                        break;
170                        }
171        } while (true);
172     
173        (void)sendall(client, content, sizeof(content));
```

rveerama1, 2019-11-21:
False positive.

closed 2019-11-21 by rveerama1