zephyrproject-rtos/zephyr #20510: [Coverity CID :205660]Incorrect expression in /subsys/debug/tracing/ctf/ctf_top.c

aasthagr, 2019-11-08:
Static code scan issues seen in File: /subsys/debug/tracing/ctf/ctf_top.c
 Category: Incorrect expression
 Function: sys_trace_thread_name_set
 Component: Other
 CID: 205660
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aasthagr, 2019-11-08:
```
*** CID 205660:  Incorrect expression  (NO_EFFECT)
/subsys/debug/tracing/ctf/ctf_top.c: 101 in sys_trace_thread_name_set()
95     
96     void sys_trace_thread_name_set(struct k_thread *thread)
97     {
98     #if defined(CONFIG_THREAD_NAME)
99          ctf_bounded_string_t name = { "Unnamed thread" };
100     
     CID 205660:  Incorrect expression  (NO_EFFECT)
     Comparing an array to null is not useful: "thread->name != NULL", since the test will always evaluate as true.
101        if (thread->name != NULL) {
102                        strncpy(name.buf, thread->name, sizeof(name.buf));
103                        /* strncpy may not always null-terminate */
104                        name.buf[sizeof(name.buf) - 1] = 0;
105        }
106        ctf_top_thread_name_set(
```

aescolar, 2019-11-11:
5b322d93317

aescolar, 2019-11-11:
@nslowell (You don't seem to be a "GH collaborator" in the project so I cannot assign this to you, but it does seem to originate from your commit
CC: @fnde-ot 
CC: @MaureenHelm

nslowell, 2019-11-11:
Fix looks good.  Sorry and thanks.

PR #20574 (2019-11-12): debug: tracing: Use proper API to fetch thread name
Use k_thread_name_get API to fetch thread name.

Fixes: #20509
Fixes: #20510
commit 6990a379 (2019-11-12): debug: tracing: Use proper API to fetch thread name

Use k_thread_name_get API to fetch thread name.

Fixes: #20509
Fixes: #20510
closed 2019-11-12 by ioannisg