zephyrproject-rtos/zephyr #20511: [Coverity CID :205612]Control flow issues in /subsys/disk/disk_access_spi_sdhc.c

aasthagr, 2019-11-08:
Static code scan issues seen in File: /subsys/disk/disk_access_spi_sdhc.c
 Category: Control flow issues
 Function: sdhc_spi_detect
 Component: Other
 CID: 205612
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aescolar, 2019-11-11:
```
567                if (err != 0) {
568                        /* Card never finished power-up */
   	
CID 205612 (#1 of 1): Logically dead code (DEADCODE)
dead_error_line: Execution cannot reach this statement: return -60;.
569                        return -ETIMEDOUT;
570                }
571        }
```

aescolar, 2019-11-11:
c7e625f2b39

aescolar, 2019-11-11:
@JunYangNXP

aescolar, 2019-11-11:
@MaureenHelm note that @JunYangNXP is not a GH collaborator so I could not assign him

PR #20705 (2019-11-18): disk: sdhc: Remove logically dead code from spi sdhc driver
We already returned out of the function if err is nonzero, therefore it
is impossible to reach this return statement.

Coverity-CID: 205612

Fixes #20511
closed 2019-11-18 by carlescufi