zephyrproject-rtos/zephyr #20515: [Coverity CID :205670]Code maintainability issues in /subsys/settings/src/settings_nvs.c

aasthagr, 2019-11-08:
Static code scan issues seen in File: /subsys/settings/src/settings_nvs.c
 Category: Code maintainability issues
 Function: settings_nvs_save
 Component: Other
 CID: 205670
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aasthagr, 2019-11-08:
```
*** CID 205670:  Code maintainability issues  (UNUSED_VALUE)
/subsys/settings/src/settings_nvs.c: 174 in settings_nvs_save()
168                        if (strcmp(name, rdname)) {
169                                        continue;
170                        }
171     
172                        if ((delete) && (name_id == cf->last_name_id)) {
173                                        cf->last_name_id--;
     CID 205670:  Code maintainability issues  (UNUSED_VALUE)
     Assigning value from "nvs_write(&cf->cf_nvs, 32768, &cf->last_name_id, 2U)" to "rc" here, but that stored value is overwritten before it can be used.
174                                        rc = nvs_write(&cf->cf_nvs, NVS_NAMECNT_ID,
175                                                               &cf->last_name_id, sizeof(u16_t));
176                        }
177     
178                        if (delete) {
179                                        rc = nvs_delete(&cf->cf_nvs, name_id);
```

dleach02, 2019-11-12:
The last instances of assignment to rc likely need to be replaced with:

(void) nvs_write(...)

PR #20656 (2019-11-13): setting: fix NVS error handling on write
In settings write NVS errors were not handled
in a few places.
This patch improve that.

fixes #20515
commit aae2c846 (2019-11-13): setting: fix NVS error handling on write

In settings write NVS errors were not handled
in a few places.
This patch improve that.

fixes #20515
closed 2019-11-13 by MaureenHelm