zephyrproject-rtos/zephyr #20517: [Coverity CID :205640]Control flow issues in /subsys/testsuite/ztest/src/ztest.c

aasthagr, 2019-11-08:
Static code scan issues seen in File: /subsys/testsuite/ztest/src/ztest.c
 Category: Control flow issues
 Function: z_test_1cpu_stop
 Component: Other
 CID: 205640
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aescolar, 2019-11-11:
efc1ec037be

andyross, 2019-11-11:
FWIW: the complaint is that there's a loop of the form:

```
for(int i=0; i<MACRO_VALUE; i++) {
    ...
}
```

And the test is flagging the loop interior as dead because MACRO_VALUE evaluates to "1" in the configuration being tested.  We have coverage for that case (SMP targets with more than one CPU) for that elsewhere, but coverity doesn't know about it.

What's policy here?  We can put an #if around it, but that's just replacing one kind of dead code with another that coverity likes better, and IMHO not worth it.  Is there a way to get it access to the variant configurations? (Maybe not, I know it's sort of a performance problem already).

PR #27132 (2020-07-25): Address multiple Coverity warning bugs
Multiple coverity issues are addressed in this PR.

Fixes #26985 
Fixes #26986 
Fixes #25724
Fixes #25726 
Fixes #25790 
Fixes #20516 
Fixes #20517 
Fixes #18960
commit 48f7f119 (2020-07-25): subsys/ztest: Suppress Coverity warning

Coverity warnings on dead loop code. We know this can
occur if the NUM_CPUHOLD is defined as zero which occurs
when CONFIG_SMP is false.

Fixes #20516
Fixes #20517
closed 2020-07-25 by nashif