zephyrproject-rtos/zephyr #20518: [Coverity CID :205647]Memory - illegal accesses in /tests/arch/x86/info/src/memmap.c

aasthagr, 2019-11-08:
Static code scan issues seen in File: /tests/arch/x86/info/src/memmap.c
 Category: Memory - illegal accesses
 Function: memmap
 Component: Tests
 CID: 205647
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aescolar, 2019-11-11:
2cf52476ea6

dleach02, 2020-08-15:
```
30        printk(", max %d entries\n", CONFIG_X86_MEMMAP_ENTRIES);
31
  	3. Condition i < 64, taking true branch.
  	6. Condition i < 64, taking true branch.
  	7. cond_at_most: Checking i < 64 implies that i may be up to 63 on the true branch.
32        for (int i = 0; i < CONFIG_X86_MEMMAP_ENTRIES; ++i) {
  	8. alias: Assigning: entry = x86_memmap + i. entry may now point to as high as element 63 of x86_memmap (which consists of 1 24-byte elements).
33                struct x86_memmap_entry *entry = x86_memmap + i;
34                const char *label;
35
  	4. Switch case value X86_MEMMAP_ENTRY_UNUSED.
  	
CID 205647 (#1 of 1): Out-of-bounds read (OVERRUN)
9. overrun-local: Overrunning array of 24 bytes at byte offset 1512 by dereferencing pointer entry.
36                switch (entry->type) {
37                case X86_MEMMAP_ENTRY_UNUSED:
  	5. Continuing loop.
38                        continue;
39                case X86_MEMMAP_ENTRY_RAM:
40                        label = "RAM";
```

dleach02, 2020-11-09:
Unless someone disagrees, I'm going to mark this as false positive and close it.

The x86_memmap array has a _weak definition to include three items. I'm suspecting that maybe this run of coverity was using multi-boot and mixed results from other builds??? Not sure because it believes the loop will run past 3 entries so maybe the CONFIG_X86_MEMMAP_ENTRIES is 64?

aasthagr, 2020-11-18:
Coverity calculated the size of the array incorrectly assuming the last element is not there. Its a false Positive. Closing it.

closed 2020-11-18 by aasthagr