zephyrproject-rtos/zephyr #20519: [Coverity CID :205616]Memory - corruptions in /tests/crypto/tinycrypt/src/ecc_dsa.c

aasthagr, 2019-11-08:
Static code scan issues seen in File: /tests/crypto/tinycrypt/src/ecc_dsa.c
 Category: Memory - corruptions
 Function: cavp_verify
 Component: Tests
 CID: 205616
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aescolar, 2019-11-11:
e2ad85b696e & 062489f8732
```
564        struct tc_sha256_state_struct sha256_ctx;
565
566        printf("Test #2: ECDSAvrfy ");
567        printf("NIST-p256, SHA2-256\n");
568
   	
CID 205616 (#1 of 1): Out-of-bounds access (OVERRUN)
1. overrun-buffer-val: Overrunning struct type tc_sha256_state_struct of 108 bytes by passing it to a function which accesses it at byte offset 111. [show details]
569        return vrfy_vectors(&sha256_ctx, Msg, Qx, Qy, R, S, results, 15, verbose);
570}
```

dleach02, 2020-08-14:
This is a false positive generation from when we had a pass of coverity that had part of the system compiled for 64bit and the other part for 32 bit. The structure is 108 bytes in size and the underlying library uses this size to initialize things. Part of the structure has a size_t definition which will be different in size for 64bit systems.

closed 2020-08-14 by dleach02