zephyrproject-rtos/zephyr #20529: [Coverity CID :205658]Memory - corruptions in /tests/crypto/tinycrypt/src/sha256.c

aasthagr, 2019-11-08:
Static code scan issues seen in File: /tests/crypto/tinycrypt/src/sha256.c
 Category: Memory - corruptions
 Function: test_sha256_1
 Component: Tests
 CID: 205658
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aasthagr, 2019-11-08:
```
*** CID 205658:    (OVERRUN)
/tests/crypto/tinycrypt/src/sha256.c: 73 in test_sha256_1()
67          const char *m = "abc";
68          u8_t digest[32];
69          struct tc_sha256_state_struct s;
70     
71          (void)tc_sha256_init(&s);
72          tc_sha256_update(&s, (const u8_t *)m, strlen(m));
     CID 205658:    (OVERRUN)
     Overrunning struct type tc_sha256_state_struct of 108 bytes by passing it to a function which accesses it at byte offset 111.
73          (void)tc_sha256_final(digest, &s);
74          result = check_result(1, expected, sizeof(expected),
75                                                digest, sizeof(digest), 1);
76     
77          /**TESTPOINT: Check result*/
78          zassert_false(result, "SHA256 test #1 failed.");
/tests/crypto/tinycrypt/src/sha256.c: 71 in test_sha256_1()
65                          0xb4, 0x10, 0xff, 0x61, 0xf2, 0x00, 0x15, 0xad
66          };
67          const char *m = "abc";
68          u8_t digest[32];
69          struct tc_sha256_state_struct s;
70     
     CID 205658:    (OVERRUN)
     Overrunning struct type tc_sha256_state_struct of 108 bytes by passing it to a function which accesses it at byte offset 111.
71          (void)tc_sha256_init(&s);
72          tc_sha256_update(&s, (const u8_t *)m, strlen(m));
73          (void)tc_sha256_final(digest, &s);
74          result = check_result(1, expected, sizeof(expected),
75                                                digest, sizeof(digest), 1);
76     
```

ceolin, 2019-11-21:
We have spent some time on Security WG discussing these tinycrypt issues reported by Coverity and we concluded that they are false positive.

closed 2019-11-21 by ceolin