zephyrproject-rtos/zephyr #20530: [Coverity CID :205663]Memory - corruptions in /tests/crypto/tinycrypt/src/sha256.c

aasthagr, 2019-11-08:
Static code scan issues seen in File: /tests/crypto/tinycrypt/src/sha256.c
 Category: Memory - corruptions
 Function: test_sha256_6
 Component: Tests
 CID: 205663
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aasthagr, 2019-11-08:
```
*** CID 205663:    (OVERRUN)
/tests/crypto/tinycrypt/src/sha256.c: 218 in test_sha256_6()
212        struct tc_sha256_state_struct s;
213     
214        (void)memset(m, 0x00, sizeof(m));
215     
216        (void)tc_sha256_init(&s);
217        tc_sha256_update(&s, m, sizeof(m));
     CID 205663:    (OVERRUN)
     Overrunning struct type tc_sha256_state_struct of 108 bytes by passing it to a function which accesses it at byte offset 111.
218        (void)tc_sha256_final(digest, &s);
219     
220        result = check_result(6, expected, sizeof(expected),
221                                              digest, sizeof(digest), 1);
222     
223        /**TESTPOINT: Check result*/
/tests/crypto/tinycrypt/src/sha256.c: 216 in test_sha256_6()
210        u8_t m[56];
211        u8_t digest[32];
212        struct tc_sha256_state_struct s;
213     
214        (void)memset(m, 0x00, sizeof(m));
215     
     CID 205663:    (OVERRUN)
     Overrunning struct type tc_sha256_state_struct of 108 bytes by passing it to a function which accesses it at byte offset 111.
216        (void)tc_sha256_init(&s);
217        tc_sha256_update(&s, m, sizeof(m));
218        (void)tc_sha256_final(digest, &s);
219     
220        result = check_result(6, expected, sizeof(expected),
221                                              digest, sizeof(digest), 1);
```

ceolin, 2019-11-21:
We have spent some time on Security WG discussing these tinycrypt issues reported by Coverity and we concluded that they are false positive. It seems that coverity is getting confused with the way we running it with sanitycheck and multiple architectures.

closed 2019-11-21 by ceolin