zephyrproject-rtos/zephyr #20531: [Coverity CID :205666]Memory - corruptions in /tests/crypto/tinycrypt/src/sha256.c

aasthagr, 2019-11-08:
Static code scan issues seen in File: /tests/crypto/tinycrypt/src/sha256.c
 Category: Memory - corruptions
 Function: test_sha256_4
 Component: Tests
 CID: 205666
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aasthagr, 2019-11-08:
```
*** CID 205666:    (OVERRUN)
/tests/crypto/tinycrypt/src/sha256.c: 156 in test_sha256_4()
150                        0xbd, 0x56, 0xc6, 0x1c, 0xcc, 0xcd, 0x95, 0x04
151        };
152        const u8_t m[4] = { 0xc9, 0x8c, 0x8e, 0x55 };
153        u8_t digest[32];
154        struct tc_sha256_state_struct s;
155     
     CID 205666:    (OVERRUN)
     Overrunning struct type tc_sha256_state_struct of 108 bytes by passing it to a function which accesses it at byte offset 111.
156        (void)tc_sha256_init(&s);
157        tc_sha256_update(&s, m, sizeof(m));
158        (void)tc_sha256_final(digest, &s);
159     
160        result = check_result(4, expected, sizeof(expected),
161                                              digest, sizeof(digest), 1);
/tests/crypto/tinycrypt/src/sha256.c: 158 in test_sha256_4()
152        const u8_t m[4] = { 0xc9, 0x8c, 0x8e, 0x55 };
153        u8_t digest[32];
154        struct tc_sha256_state_struct s;
155     
156        (void)tc_sha256_init(&s);
157        tc_sha256_update(&s, m, sizeof(m));
     CID 205666:    (OVERRUN)
     Overrunning struct type tc_sha256_state_struct of 108 bytes by passing it to a function which accesses it at byte offset 111.
158        (void)tc_sha256_final(digest, &s);
159     
160        result = check_result(4, expected, sizeof(expected),
161                                              digest, sizeof(digest), 1);
162     
163        /**TESTPOINT: Check result*/
```

ceolin, 2019-11-21:
We have spent some time on Security WG discussing these tinycrypt issues reported by Coverity and we concluded that they are false positive. It seems that coverity is getting confused with the way we running it with sanitycheck and multiple architectures.

closed 2019-11-21 by ceolin