zephyrproject-rtos/zephyr #20532: [Coverity CID :205674]Memory - corruptions in /tests/crypto/tinycrypt/src/sha256.c

aasthagr, 2019-11-08:
Static code scan issues seen in File: /tests/crypto/tinycrypt/src/sha256.c
 Category: Memory - corruptions
 Function: test_sha256_3
 Component: Tests
 CID: 205674
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aasthagr, 2019-11-08:
```
*** CID 205674:    (OVERRUN)
/tests/crypto/tinycrypt/src/sha256.c: 130 in test_sha256_3()
124        const u8_t m[1] = { 0xbd };
125        u8_t digest[32];
126        struct tc_sha256_state_struct s;
127     
128        (void)tc_sha256_init(&s);
129        tc_sha256_update(&s, m, sizeof(m));
     CID 205674:    (OVERRUN)
     Overrunning struct type tc_sha256_state_struct of 108 bytes by passing it to a function which accesses it at byte offset 111.
130        (void)tc_sha256_final(digest, &s);
131     
132        result = check_result(3, expected, sizeof(expected),
133                                              digest, sizeof(digest), 1);
134     
135        /**TESTPOINT: Check result*/
/tests/crypto/tinycrypt/src/sha256.c: 128 in test_sha256_3()
122                        0xe1, 0x12, 0x04, 0xc0, 0x8f, 0xfe, 0x73, 0x2b
123        };
124        const u8_t m[1] = { 0xbd };
125        u8_t digest[32];
126        struct tc_sha256_state_struct s;
127     
     CID 205674:    (OVERRUN)
     Overrunning struct type tc_sha256_state_struct of 108 bytes by passing it to a function which accesses it at byte offset 111.
128        (void)tc_sha256_init(&s);
129        tc_sha256_update(&s, m, sizeof(m));
130        (void)tc_sha256_final(digest, &s);
131     
132        result = check_result(3, expected, sizeof(expected),
133                                              digest, sizeof(digest), 1);
```

ceolin, 2019-11-21:
We have spent some time on Security WG discussing these tinycrypt issues reported by Coverity and we concluded that they are false positive. It seems that coverity is getting confused with the way we running it with sanitycheck and multiple architectures.

closed 2019-11-21 by ceolin