zephyrproject-rtos/zephyr #20535: [Coverity CID :205619]Null pointer dereferences in /tests/net/ieee802154/fragment/src/main.c

aasthagr, 2019-11-08:
Static code scan issues seen in File: /tests/net/ieee802154/fragment/src/main.c
 Category: Null pointer dereferences
 Function: test_fragment
 Component: Tests
 CID: 205619
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aescolar, 2019-11-09:
@jukkar please triage further as needed

aescolar, 2019-11-11:
```
438static bool test_fragment(struct net_fragment_data *data)
439{
   	1. assign_zero: Assigning: rxpkt = NULL.
440        struct net_pkt *rxpkt = NULL;
441        struct net_pkt *f_pkt = NULL;
442        int result = false;
443        struct ieee802154_fragment_ctx ctx;
444        struct net_buf *buf, *dfrag;
445        struct net_pkt *pkt;
446        int hdr_diff;
447
448        pkt = create_pkt(data);
   	2. Condition !pkt, taking false branch.
449        if (!pkt) {
450                TC_PRINT("%s: failed to create buffer\n", __func__);
451                goto end;
452        }
453
454#if DEBUG > 0
455        printk("length before compression %zd\n", net_pkt_get_len(pkt));
456        net_pkt_hexdump(pkt, "before-compression");
457#endif
458
459        hdr_diff = net_6lo_compress(pkt, data->iphc);
   	3. Condition hdr_diff < 0, taking false branch.
460        if (hdr_diff < 0) {
461                TC_PRINT("compression failed\n");
462                goto end;
463        }
464
   	4. Condition !ieee802154_fragment_is_needed(pkt, 0), taking false branch.
465        if (!ieee802154_fragment_is_needed(pkt, 0)) {
466                f_pkt = pkt;
467                pkt = NULL;
468
469                goto reassemble;
470        }
471
472        f_pkt = net_pkt_alloc(K_FOREVER);
   	5. Condition !f_pkt, taking false branch.
473        if (!f_pkt) {
474                goto end;
475        }
476
477        ieee802154_fragment_ctx_init(&ctx, pkt, hdr_diff, data->iphc);
478        frame_buf.len = 0U;
479
480        buf = pkt->buffer;
   	6. Condition buf, taking true branch.
   	9. Condition buf, taking false branch.
481        while (buf) {
482                ieee802154_fragment(&ctx, &frame_buf, data->iphc);
483                buf = ctx.buf;
484
485                dfrag = net_pkt_get_frag(f_pkt, K_FOREVER);
   	7. Condition !dfrag, taking false branch.
486                if (!dfrag) {
487                        goto end;
488                }
489
490                memcpy(dfrag->data, frame_buf.data, frame_buf.len);
491                dfrag->len = frame_buf.len;
492
493                net_pkt_frag_add(f_pkt, dfrag);
494
495                frame_buf.len = 0U;
   	8. Jumping back to the beginning of the loop.
496        }
497
498        net_pkt_unref(pkt);
499        pkt = NULL;
500
501reassemble:
502
503#if DEBUG > 0
504        printk("length after compression and fragmentation %zd\n",
505               net_pkt_get_len(f_pkt));
506        net_pkt_hexdump(f_pkt, "after-compression");
507#endif
508
509        buf = f_pkt->buffer;
   	10. Condition buf, taking false branch.
510        while (buf) {
511                rxpkt = net_pkt_rx_alloc(K_FOREVER);
512                if (!rxpkt) {
513                        goto end;
514                }
515
516                dfrag = net_pkt_get_frag(rxpkt, K_FOREVER);
517                if (!dfrag) {
518                        goto end;
519                }
520
521                memcpy(dfrag->data, buf->data, buf->len);
522                dfrag->len = buf->len;
523
524                net_pkt_frag_add(rxpkt, dfrag);
525
526                net_pkt_set_overwrite(rxpkt, true);
527
528                switch (ieee802154_reassemble(rxpkt)) {
529                case NET_OK:
530                        buf = buf->frags;
531                        break;
532                case NET_CONTINUE:
533                        goto compare;
534                case NET_DROP:
535                        net_pkt_unref(rxpkt);
536                        goto end;
537                }
538        }
539
540compare:
541#if DEBUG > 0
542        printk("length after reassembly and uncompression %zd\n",
543               net_pkt_get_len(rxpkt));
544        net_pkt_hexdump(rxpkt, "after-uncompression");
545#endif
546
   	
CID 205619 (#1 of 1): Explicit null dereferenced (FORWARD_NULL)
11. var_deref_model: Passing null pointer rxpkt to compare_data, which dereferences it. [show details]
547        if (compare_data(rxpkt, data)) {
548                result = true;
549        }
````

tbursztyka, 2019-12-03:
false positive

closed 2019-12-03 by tbursztyka