zephyrproject-rtos/zephyr #20538: [Coverity CID :205673]Memory - corruptions in /tests/net/ppp/driver/src/main.c

aasthagr, 2019-11-08:
Static code scan issues seen in File: /tests/net/ppp/driver/src/main.c
 Category: Memory - corruptions
 Function: ppp_verify_fcs_1
 Component: Tests
 CID: 205673
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aasthagr, 2019-11-08:
```
*** CID 205673:  Memory - corruptions  (OVERRUN)
/tests/net/ppp/driver/src/main.c: 357 in ppp_verify_fcs_1()
351     
352        net_pkt_unref(pkt);
353     }
354     
355     static void ppp_verify_fcs_1(void)
356     {
     CID 205673:  Memory - corruptions  (OVERRUN)
     Overrunning array "ppp_recv_data1" of 45 bytes by passing it to a function which accesses it at byte offset 45 using argument "45".
357        ppp_verify_fcs(ppp_recv_data1, sizeof(ppp_recv_data1));
358     }
359     
360     static bool calc_fcs(struct net_pkt *pkt, u16_t *fcs)
361     {
362        struct net_buf *buf;
```

jukkar, 2019-11-26:
False positive

closed 2019-11-26 by jukkar