zephyrproject-rtos/zephyr #20539: [Coverity CID :205637]Resource leaks in /tests/net/socket/tcp/src/main.c

aasthagr, 2019-11-08:
Static code scan issues seen in File: /tests/net/socket/tcp/src/main.c
 Category: Resource leaks
 Function: test_v4_accept_timeout
 Component: Tests
 CID: 205637
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aescolar, 2019-11-11:
8e6aaaf919d 
```
422void test_v4_accept_timeout(void)
423{
424        /* Test if accept() will timeout properly */
425        int s_sock;
426        int new_sock;
427        u32_t tstamp;
428        struct sockaddr_in s_saddr;
429        struct sockaddr addr;
430        socklen_t addrlen = sizeof(addr);
431
432        prepare_sock_tcp_v4(CONFIG_NET_CONFIG_MY_IPV4_ADDR, SERVER_PORT,
433                            &s_sock, &s_saddr);
434
435        test_bind(s_sock, (struct sockaddr *)&s_saddr, sizeof(s_saddr));
436        test_listen(s_sock);
437
438        test_fcntl(s_sock, F_SETFL, O_NONBLOCK);
439
440        tstamp = k_uptime_get_32();
   	1. open_arg: test_accept_timeout opens handle stored into new_sock. [show details]
441        test_accept_timeout(s_sock, &new_sock, &addr, &addrlen);
   	2. Condition k_uptime_get_32() - tstamp <= 100, taking true branch.
   	3. Condition k_uptime_get_32() - tstamp <= 100, taking true branch.
   	4. Condition "", taking true branch.
   	5. Condition "", taking true branch.
442        zassert_true(k_uptime_get_32() - tstamp <= 100, "");
443
444        test_close(s_sock);
445
446        k_sleep(TCP_TEARDOWN_TIMEOUT);
   	
CID 205637 (#1 of 1): Resource leak (RESOURCE_LEAK)
6. leaked_handle: Handle variable new_sock going out of scope leaks the handle.
447}
```

rveerama1, 2019-11-21:
False positive. If new_sock accepted it is error case. Test will fail. No need to close the socket.

closed 2019-11-21 by rveerama1