zephyrproject-rtos/zephyr #20540: [Coverity CID :205656]Error handling issues in /tests/net/tcp/src/main.c

aasthagr, 2019-11-08:
Static code scan issues seen in File: /tests/net/tcp/src/main.c
 Category: Error handling issues
 Function: setup_ipv4_tcp
 Component: Tests
 CID: 205656
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aasthagr, 2019-11-08:
```
*** CID 205656:  Error handling issues  (CHECKED_RETURN)
/tests/net/tcp/src/main.c: 385 in setup_ipv4_tcp()
379        pkt = net_pkt_alloc_with_buffer(iface, sizeof(data),
380                                                                        AF_INET, IPPROTO_TCP, K_FOREVER);
381        if (!pkt) {
382                        return NULL;
383        }
384     
     CID 205656:  Error handling issues  (CHECKED_RETURN)
     Calling "net_ipv4_create" without checking return value (as is done elsewhere 8 out of 10 times).
385        net_ipv4_create(pkt, remote_addr, local_addr);
386     
387        tcp_hdr.src_port = htons(remote_port);
388        tcp_hdr.dst_port = htons(local_port);
389     
390        net_pkt_write(pkt, &tcp_hdr, sizeof(struct net_tcp_hdr));
```

aescolar, 2019-11-11:
CC @tbursztyka

PR #20893 (2019-11-22): Fix TCP tests Coverity issue
Check return values of function.

Fixes #20540
Coverity CID :205656
commit 94e04855 (2019-11-22): tests: net: Fix TCP tests Coverity issue

Check return values of function.

Fixes #20540
Coverity CID :205656
closed 2019-11-22 by nashif