zephyrproject-rtos/zephyr #20545: imgtool: signing image fails: missing DT_FLASH_WRITE_BLOCK_SIZE

LeBlue, 2019-11-09:
**Describe the bug**
The image signing with imgtool (1.4.0) fails either through west or directly

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b nrf52840_pca10056 -d build/sign_example zephyr/samples/subsys/mgmt/mcumgr/smp_svr
2. west sign -t imgtool -d build/sign_example -- --key SIGNING_KEY.pem
produces:
```
ERROR: imgtool parameter unknown, build directory has no device tree define DT_FLASH_WRITE_BLOCK_SIZE
```

**Environment:**
 - zephyr https://github.com/zephyrproject-rtos/zephyr/commit/80faac41bc6106ab3c679c05c45d28766258a1d1
- imgtool 1.4.0

**Additional context**
commit https://github.com/zephyrproject-rtos/zephyr/commit/9da1d41a12c114a596a9413b84638b08013f6317 drops loading device tree config needed for building the sign command line in https://github.com/zephyrproject-rtos/zephyr/blob/80faac41bc6106ab3c679c05c45d28766258a1d1/scripts/west_commands/sign.py#L208

mbolivar, 2019-11-12:
@galak .... I guess we found an issue where a script needs DT information :)

@ulfalize @galak what's the right thing to do here? Can I import dtlib or edtlib and get the value of DT_FLASH_WRITE_BLOCK_SIZE from there somehow? If not, I may need to parse the .conf again.

galak, 2019-11-12:
> @galak .... I guess we found an issue where a script needs DT information :)
> 
> @ulfalizer @galak what's the right thing to do here? Can I import dtlib or edtlib and get the value of DT_FLASH_WRITE_BLOCK_SIZE from there somehow? If not, I may need to parse the .conf again.

Yeah, we should import edtlib and pull the info out from there.

mbolivar, 2019-11-12:
ACK. For those watching, the regression was introduced in 9da1d41a12c114a596a9413b84638b08013f6317. Working on a fix that doesn't require reverting that, but if you want to get unblocked you can just revert that for now.

PR #20645 (2019-11-15): West sign with edtlib
```
scripts: west sign: use edtlib for flash configuration

This command needs access to DT configuration, but can no longer
access it through BuildConfiguration since
9da1d41.

Import edtlib and use that instead.
```

cc @LeBlue

Fixes: #20545
commit d8f459ab (2019-11-15): scripts: west sign: use edtlib for flash configuration

This command needs access to DT configuration, but can no longer
access it through BuildConfiguration since
9da1d41a12c114a596a9413b84638b08013f6317.

Import edtlib and use that instead.

Fix up some other error handling and output issues while we're here to
make the script's behavior easier to inspect and debug.

Fixes: #20545
closed 2019-11-15 by galak