zephyrproject-rtos/zephyr #20561: Crypto API: Separate IV from ciphertext when based on struct cipher_ctx::flags

cbsiddharth, 2019-11-10:
**Problem:**
The crypto API now prefixes the IV to to the ciphertext. This is because Tinycrypt expects the IV and ciphertext to be in the contiguous memory for better performance. This is a problem if applications what to do inplace encrypt/decrypt (`in_buf == out_buf`) as now, the `in_buf` has to be larger by 2x block length and the IV has to be removed manually. Besides, not all use-cases expect IV to be prefixed to the ciphertext.

**Proposed Solution:**
Create new flag and pass it in struct `cipher_ctx::flags` that would allow `out_buf` to have only ciphertext. This means the TC shim would now have to do some extra buffer copy but MbedTLS can skip some.

I will raise a PR based on comments here.

cbsiddharth, 2019-11-10:
Related to: #20434

@tbursztyka @chrta @galak @nashif

cbsiddharth, 2019-11-20:
Does anyone have any objections/suggestions on this approach? Do I go ahead to a PR?

tbursztyka, 2019-11-20:
Go with an RFC PR.

cbsiddharth, 2019-11-27:
@tbursztyka: I've raised PR. Could you please help add the RFC label to it? I cannot add it.

tbursztyka, 2019-11-28:
it was just needed in the title of the PR

PR #21044 (2020-01-16): crypto: Add support for non-IV-prefixed CBC ciphers in mbedTLS shim
This patch adds a new capability flag `CAP_NO_IV_PREFIX` that allows CBC operation without prefixing the IV to the output ciphertext.

When `CAP_NO_IV_PREFIX` is active, the IV passed to cipher_cbc_op() is not modified. So the length of the ciphertext block is equal to the plaintext block, allowing in-place encryption/decryption.

This flag is useful when the IV is derived from other sources (for instance a chained MAC serving as IV) and not directly transmitted to the other end.

**Notes:**
* Fixes: #20561
* Required for: #19 [... cut]
closed 2020-01-16 by carlescufi