zephyrproject-rtos/zephyr #20571: devicetree: fix non-deterministic multi-level interrupt encodings

pabigot, 2019-11-11:
The scripting that generates multi-level IRQ information does not guarantee determininstic instance order.  While it can be patched by hard-coding specific instances as with #20550 and #20562 the problem needs to be solved more generally.

Example output from #20558.

**Screenshots or console output**
```
[184/188] Generating linker_pass_final.cmd
[185/188] Generating isr_tables.c
FAILED: cd /home/buildslave/src/github.com/zephyrproject-rtos/zephyr/sanity-out/rv32m1_vega_ri5cy/samples/bluetooth/peripheral_hr/sample.bluetooth.peripheral_hr_rv32m1_vega_ri5cy/zephyr && /opt/sdk/zephyr-sdk-0.10.3/riscv64-zephyr-elf/bin/riscv64-zephyr-elf-objcopy --input-target=elf32-littleriscv --output-target=binary --only-section=.intList /home/buildslave/src/github.com/zephyrproject-rtos/zephyr/sanity-out/rv32m1_vega_ri5cy/samples/bluetooth/peripheral_hr/sample.bluetoot.h.peripheral_hr_rv32m1_vega_ri5cy/zephyr/zephyr_prebuilt.elf isrList.bin && /usr/bin/python3 /home/buildslave/src/github.com/zephyrproject-rtos/zephyr/arch/common/gen_isr_tables.py --output-source isr_tables.c --kernel /home/buildslave/src/github.com/zephyrproject-rtos/zephyr/sanity-out/rv32m1_vega_ri5cy/samples/bluetooth/peripheral_hr/sample.bluetooth.peripheral_hr_rv32m1_vega_ri5cy/zephyr/zephyr_prebuilt.elf --intlist isrList.bin   --sw-isr-table
gen_isr_tables.py: multiple registrations at table_index 25 for irq 25 (0x19)
Traceback (most recent call last):
  File "/home/buildslave/src/github.com/zephyrproject-rtos/zephyr/arch/common/gen_isr_tables.py", line 314, in <module>
    main()
  File "/home/buildslave/src/github.com/zephyrproject-rtos/zephyr/arch/common/gen_isr_tables.py", line 306, in main
    error("multiple registrations at table_index %d for irq %d (0x%x)" % (table_index, irq, irq))
  File "/home/buildslave/src/github.com/zephyrproject-rtos/zephyr/arch/common/gen_isr_tables.py", line 36, in error
    raise Exception()
Exception
ninja: build stopped: subcommand failed.
```

carlescufi, 2019-11-11:
Setting priority to high because this completely defeats the deterministic nature of DT.

ulfalizer, 2019-11-12:
Didn't notice this one until now, but I wonder if https://github.com/zephyrproject-rtos/zephyr/pull/20597 solves it. In that case it was only an issue on Python 3.5 and earlier. CI runs 3.5.

PR #20597 (2019-11-12): scripts: dtlib/edtlib: Make output consistent on Python 3.5-
Change:

```
scripts: dtlib/edtlib: Make output consistent on Python 3.5-

Python 3.5 and earlier do not preserve dictionary insertion order when
iterating over dictionaries, and do not give the same order between
runs. This broke the dtlib and edtlib test suites and made the output
jump around randomly between runs. It also made device INST_<n> numbers
non-deterministic, which broke some code on Python 3.5 (though
hardcoding device instance numbers in the code might be a bit shaky).

Fix it by using collections.OrderedDict instead o [... cut]
commit 72158858 (2019-11-12): scripts: dtlib/edtlib: Make output consistent on Python 3.5-

Python 3.5 and earlier do not preserve dictionary insertion order when
iterating over dictionaries, and do not give the same order between
runs. This broke the dtlib and edtlib test suites and made the output
jump around randomly between runs. It also made device INST_<n> numbers
non-deterministic, which broke some code on Python 3.5 (though
hardcoding device instance numbers in the code might be a bit shaky).

Fix it by using collections.OrderedDict instead of plain dict wherever
order matters. This makes the output identical on al [... cut]
closed 2019-11-12 by galak