zephyrproject-rtos/zephyr #20576: DTS overlay files must include full path name

pfl, 2019-11-11:
CMake handles overlay files inconsistently. My sample directory contains an overlay file:
-rw-rw-r-- 1 pflykt pflykt 204 Nov  8 15:11 frdm_uart2_dts.overlay

But when I run cmake, cc1 does not find the DTS overlay file:
```
(samples_fona808)$ cmake -B build -GNinja -DBOARD=frdm_k64f -DDTC_OVERLAY_FILE=frdm_uart2_dts.overlay
Zephyr version: 2.0.99
-- Selected BOARD frdm_k64f
-- Found west: /home/pflykt/.local/bin/west (found suitable version "0.6.0", minimum required is "0.6.0")
-- Loading /home/pflykt/projects/zephyr/boards/arm/frdm_k64f/frdm_k64f.dts as base
-- Overlaying frdm_uart2_dts.overlay
cc1: fatal error: frdm_uart2_dts.overlay: No such file or directory
compilation terminated.
CMake Error at /home/pflykt/projects/zephyr/cmake/dts.cmake:128 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  /home/pflykt/projects/zephyr/cmake/app/boilerplate.cmake:460 (include)
  CMakeLists.txt:7 (include)
```
Adding the full path to the command succeeds:
```
samples_fona808)$ cmake -B build -GNinja -DBOARD=frdm_k64f -DDTC_OVERLAY_FILE=$(pwd)/frdm_uart2_dts.overlay
Zephyr version: 2.0.99
-- Selected BOARD frdm_k64f
-- Found west: /home/pflykt/.local/bin/west (found suitable version "0.6.0", minimum required is "0.6.0")
-- Loading /home/pflykt/projects/zephyr/boards/arm/frdm_k64f/frdm_k64f.dts as base
-- Overlaying /home/pflykt/projects/zephyr/samples/fona808/frdm_uart2_dts.overlay
Devicetree configuration written to /home/pflykt/projects/zephyr/samples/fona808/build/zephyr/include/generated/generated_dts_board.conf
Parsing Kconfig tree in /home/pflykt/projects/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/pflykt/projects/zephyr/boards/arm/frdm_k64f/frdm_k64f_defconfig'
...
```
The same thing happens when `DTC_OVERLAY_FILE` is defined in `CMakeLists.txt`.

Documentation in https://docs.zephyrproject.org/latest/application/index.html#devicetree-overlays is very quiet about including the full path name, and the above handling of DTS overlay files shows inconsistent behavior against project overlay files. All overlay files must be handled in a consistent manner, please fix.

SebastianBoe, 2019-11-11:
CONF_FILE supports project-relative file paths so DTC_OVERLAY_FILE should to.

I see that Kconfig supports this by running kconfig with an appropriate working directory:

```
  WORKING_DIRECTORY ${APPLICATION_SOURCE_DIR}
  # The working directory is set to the app dir such that the user
  # can use relative paths in CONF_FILE, e.g. CONF_FILE=nrf5.conf

```

whereas DT uses the binary directory as the working directory. 

`    WORKING_DIRECTORY ${PROJECT_BINARY_DIR}
`

if we could change the working directory for DT without breaking backwards compatiblity or increasing the complexity significantly then we can be consistent with Kconfig and support this for DT as well.

closed 2019-12-09 by nashif

jefffhaynes, 2021-04-09:
This is still a problem

```
west build -b nrf52840dk_nrf52840 -p auto -- -DDTC_OVERLAY_FILE=./boards\nrf52840dk_nrf52840.overlay
-- west build: generating a build system
Including boilerplate (Zephyr base (cached)): C:/Users/Jeff/Source/Repos/firmware-nrf52840-5340/zephyr/cmake/app/boilerplate.cmake
-- Application: C:/Users/Jeff/Source/Repos/firmware-nrf52840-5340/zephyr/samples/sensor/fxos8700
-- Zephyr version: 2.5.99 (C:/Users/Jeff/Source/Repos/firmware-nrf52840-5340/zephyr)
-- Found west (found suitable version "0.10.1", minimum required is "0.7.1")
-- Board: nrf52840dk_nrf52840
-- Cache files will be written to: C:/Users/Jeff/Source/Repos/firmware-nrf52840-5340/zephyr/.cache
-- Found toolchain: gnuarmemb (C:/gnu_arm_embedded)
-- Found BOARD.dts: C:/Users/Jeff/Source/Repos/firmware-nrf52840-5340/zephyr/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts
-- Found devicetree overlay: ./boards/nrf52840dk_nrf52840.overlay
cc1.exe: fatal error: ./boards/nrf52840dk_nrf52840.overlay: No such file or directory
compilation terminated.
CMake Error at C:/Users/Jeff/Source/Repos/firmware-nrf52840-5340/zephyr/cmake/dts.cmake:157 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  C:/Users/Jeff/Source/Repos/firmware-nrf52840-5340/zephyr/cmake/app/boilerplate.cmake:554 (include)
  C:/Users/Jeff/Source/Repos/firmware-nrf52840-5340/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  C:/Users/Jeff/Source/Repos/firmware-nrf52840-5340/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:40 (include_boilerplate)
  CMakeLists.txt:4 (find_package)
```

skandaprasad, 2021-08-17:
@jefffhaynes 
Can you please let me know if you found a solution for your issue about not finding the overlay file even though it exists?

Michal-Pogorzelec, 2023-12-14:
The issue still exists - anyone know how to solve it?
[image]

BujSet, 2025-06-03:
I ran into this issue today on my windows machine for the esp32s3_devkitc board, and the error message seemed to indicate that it was a file pathing issue. I.e. in addition to providing the full path, it must be specified in Unix style pathing. `$(pwd)` expands to windows style pathing unless configured otherwise. @Michal-Pogorzelec if this is still an issue for you, could you try: 

```
west build -b nucleo_f411re samples/basic/blinky -DDTC_OVERLAY_FILE="C:/Users/micha/zephyrproject/zephyr/myoverlay.overlay"
```

Doing this seemed to resolve the `Found devicetree overlay` but `fatal error ... No such file or directory` error for me with the esp32s3 board.

PR #20578 (2019-12-09): cmake: Change the DTS preprocessing work directory
Change the DTS preprocessor working directory from the binary
directory to the application directory.

This is done so that the user can specify
-DDTC_OVERLAY_FILE=overlay.dts with a relative path from the
application directory as is possible for CONF_FILE, and as is
reasonably expected to be possible by users.

This fixes #20576