zephyrproject-rtos/zephyr #20582: samples/subsys/logging/syst is broken when building with gcc-arm-none-eabi-7-2018-q2-update

pfalcon, 2019-11-11:
At Linaro, we perform CI builds with GNU Arm Embedded toolchain (`ZEPHYR_TOOLCHAIN_VARIANT=gccarmemb`, https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads) in addition to Zephyr SDK. Since https://github.com/zephyrproject-rtos/zephyr/pull/19637 landed, the sample samples/subsys/logging/syst fails when built with gccarmemb version we use, `gcc-arm-none-eabi-7-2018-q2-update`. Example of a failure is at https://ci.linaro.org/job/zephyr-upstream/PLATFORM=mps2_an385,ZEPHYR_TOOLCHAIN_VARIANT=gnuarmemb,label=docker-xenial-amd64-13/4522/console . Exact compile error is:

~~~
00:59:23.801 In file included from /home/buildslave/srv/toolchain/gcc-arm-none-eabi-7-2018-q2-update/arm-none-eabi/include/wchar.h:336:0,
00:59:23.801                  from /home/buildslave/workspace/zephyr-upstream/modules/debug/mipi-sys-t/library/src/mipi_syst_api.c:606:
00:59:23.801 /home/buildslave/srv/toolchain/gcc-arm-none-eabi-7-2018-q2-update/arm-none-eabi/include/ssp/wchar.h:78:67: error: unknown type name 'FILE'; did you mean '__FILE'?
00:59:23.801  __ssp_decl(wchar_t *, fgetws, (wchar_t *__restrict __buf, int __wlen, FILE *__restrict __fp))
00:59:23.801                                                                    ^~~~
00:59:23.801                                                                    __FILE
00:59:23.801 /home/buildslave/srv/toolchain/gcc-arm-none-eabi-7-2018-q2-update/arm-none-eabi/include/ssp/wchar.h:78:234: error: unknown type name 'FILE'; did you mean '__FILE'?
00:59:23.801  __ssp_decl(wchar_t *, fgetws, (wchar_t *__restrict __buf, int __wlen, FILE *__restrict __fp))
00:59:23.801                                                                                                                                                                                                                                           ^   
00:59:23.801                                                                                                                                                                                                                                           __FILE
00:59:23.801 make[2]: *** [modules/mipi-sys-t/CMakeFiles/..__modules__debug__mipi-sys-t.dir/library/src/mipi_syst_api.c.obj] Error 1
~~~

While immediate build failure we see is with BOARD=mps2_an385, I verified that it happens with other board too (e.g. sam_e70_xplained), so it's toolchain-related.

pfalcon, 2019-11-11:
So, what happens there is that `include/logging/log_msg.h` includes `<string.h>`. But internal header machinery of gcc-arm-none-eabi-7-2018-q2-update additionally includes SSP (stack smashing protection) enabled headers. These headers appear to be somewhat convoluted, and one of them references stdio's `FILE*`, without making sure that `<stdio.h>` is included first. All that definitely looks like a bug in gcc-arm-none-eabi-7-2018-q2-update package.

For reference, more complete error message from a local build:

~~~
make[3]: Entering directory '/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/samples/subsys/logging/syst/outdir/mps2_an385'
[ 88%] Building C object modules/mipi-sys-t/CMakeFiles/..__modules__debug__mipi-sys-t.dir/zephyr_init.c.obj
[ 89%] Building C object modules/mipi-sys-t/CMakeFiles/..__modules__debug__mipi-sys-t.dir/library/src/mipi_syst_api.c.obj
In file included from /home/pfalcon/toolchain/gcc-arm-none-eabi-7-2018-q2-update/arm-none-eabi/include/ssp/string.h:35:0,
                 from /home/pfalcon/toolchain/gcc-arm-none-eabi-7-2018-q2-update/arm-none-eabi/include/string.h:180,
                 from /home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/include/logging/log_msg.h:11,
                 from /home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/include/logging/log_output.h:9,
                 from /home/pfalcon/projects-3rdparty/Embedded/Zephyr/modules/debug/mipi-sys-t/include/mipi_syst/platform.h:10,
                 from /home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/samples/subsys/logging/syst/outdir/mps2_an385/modules/mipi-sys-t/library/include/mipi_syst.h:602,
                 from /home/pfalcon/projects-3rdparty/Embedded/Zephyr/modules/debug/mipi-sys-t/library/src/mipi_syst_api.c:41:
/home/pfalcon/toolchain/gcc-arm-none-eabi-7-2018-q2-update/arm-none-eabi/include/ssp/wchar.h:78:1: error: unknown type name 'FILE'; did you mean '__FILE'?
 __ssp_decl(wchar_t *, fgetws, (wchar_t *__restrict __buf, int __wlen, FILE *__restrict __fp))
 ^
/home/pfalcon/toolchain/gcc-arm-none-eabi-7-2018-q2-update/arm-none-eabi/include/ssp/wchar.h:78:1: error: unknown type name 'FILE'; did you mean '__FILE'?
 __ssp_decl(wchar_t *, fgetws, (wchar_t *__restrict __buf, int __wlen, FILE *__restrict __fp))
 ^
modules/mipi-sys-t/CMakeFiles/..__modules__debug__mipi-sys-t.dir/build.make:75: recipe for target 'modules/mipi-sys-t/CMakeFiles/..__modules__debug__mipi-sys-t.dir/library/src/mipi_syst_api.c.obj' failed
make[3]: *** [modules/mipi-sys-t/CMakeFiles/..__modules__debug__mipi-sys-t.dir/library/src/mipi_syst_api.c.obj] Error 1
make[3]: Leaving directory '/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/samples/subsys/logging/syst/outdir/mps2_an385'
~~~

pfalcon, 2019-11-11:
There're 3 ways we could address this issue:

1. Blacklist gccarmemb from that sample. This is of course a blunt workaround of the issue.
2. Following patch would work around the issue on Zephyr side:
~~~
--- a/include/logging/log_msg.h
+++ b/include/logging/log_msg.h
@@ -7,6 +7,7 @@
 #define ZEPHYR_INCLUDE_LOGGING_LOG_MSG_H_
 
 #include <sys/atomic.h>
+#include <stdio.h>
 #include <string.h>
~~~
Of course, it's still a workaround (although a "positive", instead of "negative"), and not exactly pretty.
3. Finally, I tried a newer gccarmemb, and found that gcc-arm-none-eabi-8-2019-q3-update is devoid of this issue.

Obviously, p.3, upgrading to a newer toolchain, would be the best solution. @galak, are there reasons we stick with gcc-arm-none-eabi-7-2018-q2-update? If not, let's upgrade. Assigning to you for decision, once it's made, feel free to assign back to me for processing. Thanks.

galak, 2019-11-12:
We should update the toolchain we build with to the latest release (9-2019q4-major).

I'm also ok if you want to submit a PR with a fix to `include <stdio.h>`.

pfalcon, 2019-11-12:
> We should update the toolchain we build with to the latest release (9-2019q4-major).

Ughh, I wanted to avoid that, as then we're getting ahead of Zephyr SDK in GCC version (SDK 0.10.3 uses GCC 8.3.0). We could expect some issues when jumping to GCC 9, I guess. Anyway, downloading it to try locally a bit, then will prepare updates for our CI.

galak, 2019-11-12:
> Ughh, I wanted to avoid that, as then we're getting ahead of Zephyr SDK in GCC version (SDK 0.10.3 uses GCC 8.3.0). We could expect some issues when jumping to GCC 9, I guess. Anyway, downloading it to try locally a bit, then will prepare updates for our CI.

I know, I've already been looking at gcc9 w/ SDK 0.11.x so we hopefully shouldn't see any new issues.

nashif, 2019-11-13:
@wentongwu fyi

pfalcon, 2019-11-14:
> We could expect some issues when jumping to GCC 9, I guess.

Well (https://ci.linaro.org/view/lite-iot-ci/job/zephyr-upstream/4539/PLATFORM=mps2_an385,ZEPHYR_TOOLCHAIN_VARIANT=gnuarmemb,label=docker-xenial-amd64-13/console):

~~~
00:57:04.097 /home/buildslave/workspace/zephyr-upstream/zephyr/include/arch/arm/syscall.h:52:2: error: listing the stack pointer register 'sp' in a clobber list is deprecated [-Werror=deprecated]
00:57:04.097    52 |  __asm__ volatile("svc %[svid]\n"
00:57:04.097       |  ^~~~~~~
00:57:04.097 /home/buildslave/workspace/zephyr-upstream/zephyr/include/arch/arm/syscall.h:52:2: note: the value of the stack pointer after an 'asm' statement must be the same as it was before the statement
~~~

I'm going to try to 8-2019-q3-update next.

pfalcon, 2019-11-15:
Ok, with 8-2019-q3-update we have a green build: https://ci.linaro.org/view/lite-iot-ci/job/zephyr-upstream/4542/

> I'm also ok if you want to submit a PR with a fix to include <stdio.h>.

As mentioned above, that wouldn't be a fix, but quite an unclean (though perhaps harmless) workaround. If I were a maintainer of that header, I would oppose such workaround, at least until due diligence on a proper fix was made. Given that the solution is found, I don't try to submit questionable workarounds.

Closing this, issues of the GCC 9 warning is offloaded to https://github.com/zephyrproject-rtos/zephyr/issues/20726

closed 2019-11-15 by pfalcon