zephyrproject-rtos/zephyr #20585: z_clock_announce starvation with timeslicing active

pabigot, 2019-11-11:
**tl;dr**: the problem below has fixes but it seems likely that the same problem occurs with other timer systems if timeslicing can starve announcements on them.  The timeslicing infrastructure should be assessed to determine whether only the systick clock is at risk, or whether a general solution is required.

Issue #20445 has been isolated to code where `z_clock_set_timeout()` is always invoked before the previous timeout has a chance to expire, delaying tick announcements past the point where the 600 MHz clock causes a wrap of the 32-bit counter where unannounced ticks were accumulated.

Instrumented reproducing code and a fix for one platform are in https://github.com/pabigot/zephyr/commits/issue/annstarve which is based on the most recent version of #15487.

Below are descriptions of commits in that branch that (1) provide diagnostics of the failure, and (2) mitigate it for the Cortex-M systick timer driver.  

## DNM: tests/kernel/critical: instrument for diagnosis

Instrumentation of tests/kernel/critical demonstrates that unannounced ticks accumulated within the cortex-m-systick driver overflow the cycle counter, causing time to go backwards, preventing the loop termination condition from being reached:

```
***** Booting Zephyr OS build zephyr-v2.0.0-1961-g2633054023f5 *****
Running test suite kernel_critical
===================================================================
starting test - test_critical
Start reg1 at 18
End reg1 at 5018
Start alt1 at 5019
End alt1 at 10019
Enable timeslicing at 10020  // TS On
Start reg2 at 10023
End reg2 at 15023
Start alt2 at 15024
Time went backwards: 10013 < 17171  // CYCLE_WRAP
Time went backwards: 10013 < 17171
Time went backwards: 10013 < 17171
Time went backwards: 10013 < 17171
Time went backwards: 10013 < 17171
Time went backwards: 10013 < 17171
```

Between the line marked "TS On" and the line marked "CYCLE_WRAP" timeouts have been set before the previous timeout elapsed.  This causes the cortex_m_systick driver to accumulate unannounced cycles. At 7.158 s the 600 MHz cycle count will wrap the 32-bit internal cycle counter, causing z_clock_elapsed() to overflow and the calculated uptime to go backwards.

## drivers: systick: avoid starving clock announcements

In this commit the driver has been updated to detect that 2^31 ticks have occurred without an announcement.  In this situation the requested timeout is ignored, and instead a timeout is triggered immediately to force the announcement to be performed.

With this the instrumented output of the test on mimxrt1060_evk is below:
```
***** Booting Zephyr OS build zephyr-v2.0.0-1961-g2633054023f5 *****
Running test suite kernel_critical
===================================================================
starting test - test_critical
Start reg1 at 18
End reg1 at 5018
Start alt1 at 5019
End alt1 at 10019
Enable timeslicing at 10020
Start reg2 at 10023
End reg2 at 15023
Start alt2 at 15024
End alt2 at 20024
PASS - test_critical
===================================================================
Test suite kernel_critical succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL
```

andyross, 2020-05-11:
Is there anything actionable here?  Seems like this was just a platform bug failing to issue an announce call before wraparound?  If there's something we can reproduce in tree, we should fix it.  If not it's an enhancement request (either for another test to show the problem or for a framework to handle the overflow requirements somewhere else than in the driver, I guess?).

pabigot, 2020-05-11:
Not currently reproducible on the original platform.

closed 2020-05-11 by pabigot