zephyrproject-rtos/zephyr #20598: tests/lib/mem_alloc newlibnano target run time error

hakehuang, 2019-11-12:
**Describe the bug**
ASSERTION FAIL [(options & ((1UL << (2)))) == 0] @ ZEPHYR_BASE/kernel/thread.c:581
       Platform is capable of user mode, and test thread created with K_USER option, but neither CONFIG_TEST_USERSPACE nor CONFIG_USERSPACE is set

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=frdm_k64f ..
3. make
4. make flash
5. see log

**Expected behavior**
runs sucessfully

**Impact**
test coverage

**Screenshots or console output**

'''
***** Booting Zephyr OS version 2.1.0-rc1 *****
Running test suite test_c_lib_dynamic_memalloc
===================================================================
starting test - test_malloc
ASSERTION FAIL [(options & ((1UL << (2)))) == 0] @ ZEPHYR_BASE/kernel/thread.c:581
       Platform is capable of user mode, and test thread created with K_USER option, but neither CONFIG_TEST_USERSPACE nor CONFIG_USERSPACE is set
E: r0/a1:  0x00000004  r1/a2:  0x00000245  r2/a3:  0x00800000
E: r3/a4:  0x600095d2 r12/ip:  0x150b0000 r14/lr:  0x60006cd5
E:  xpsr:  0x61000000
E: Faulting instruction address (r15/pc): 0x60007732
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x2000017c (main)
E: Halting system

'''

**Environment (please complete the following information):**
 - OS: (Linux)
 - Toolchain (Zephyr SDK, ...)
 - Commit SHA or Version used: v2.1.0-rc1

hakehuang, 2019-11-12:
@dlech

stephanosio, 2019-11-12:
It looks like https://github.com/zephyrproject-rtos/zephyr/commit/7b1ee5cf13dc4f130b556534bd4c8dd78ceec6c6#diff-0eaa2ac33b59993522901d3fc26892b9 missed `prj_newlibnano.conf` (probably because it is not run when using Zephyr SDK due to `toolchain_whitelist: gnuarmemb`).

Adding `CONFIG_TEST_USERSPACE=y` to `prj_newlibnano.conf` should resolved this issue.

ioannisg, 2019-11-12:
there

> It looks like [7b1ee5c](https://github.com/zephyrproject-rtos/zephyr/commit/7b1ee5cf13dc4f130b556534bd4c8dd78ceec6c6) missed `prj_newlibnano.conf` (probably because it is not run when using Zephyr SDK due to `toolchain_whitelist: gnuarmemb`).
> 
> Adding `CONFIG_TEST_USERSPACE=y` to `prj_newlibnano.conf` should resolve this issue.
> 

There is some inconsistency here: the test-case for newlibnano is tagged with -userspace, but the prj_newlibnano.conf does not enable it.

Agree that CI does not catch this because it runs with zephyr toolchain...

> p.s. do we really want `CONFIG_TEST_USERSPACE` when newlibnano is used? that seems to go against what the commit referenced above is trying to solve, in which case we should not be using `ztest_user_unit_test`. cc @andrewboie

What the commit does is to require TEST_USERSPACE be enabled, if the test code defines User threads, right? So yeah, either we stop using ztest_user_unit_test if !USERSPACE in main.c or we add TEST_USERSPACE for newlibnano..

andrewboie, 2019-11-12:
If the test does stuff in user mode, CONFIG_TEST_USERSPACE needs to be enabled.

ioannisg, 2019-11-12:
> If the test does stuff in user mode, CONFIG_TEST_USERSPACE needs to be enabled.

Well, the tests is testing dynamic memory allocation, and that's coupled with user space. We should therefore run the test in user mode, imho.

andrewboie, 2019-11-12:
> We should therefore run the test in user mode, imho.

We already do.
All this is, is an omission in a testcase.yaml
I don't know why it's being proposed to not run this test in user mode

ioannisg, 2019-11-13:
I 've tried to run the test with user mode on, and it fails; I think libc globals are not put into z_libc_partition, but I might be wrong. I wonder @andrewboie if you could help a bit on this.

stephanosio, 2019-11-13:
Also it looks like we need to fix some code introduced by 472ea92e79e346ad390ebbfc1ff27a5776f4fa25.

This commit contains `toolchain_whitelist: gnuarmemb` as well as `NEWLIB_LIBC_NANO` declaration specific to `gnuarmemb` only.

Since the `arm-none-eabi` toolchain included in the Zephyr SDK also supports newlib nano, it should be whitelisted for both `gnuarmemb` and `zephyr`.

andrewboie, 2019-11-13:
> I 've tried to run the test with user mode on, and it fails; I think libc globals are not put into z_libc_partition, but I might be wrong. I wonder @andrewboie if you could help a bit on this.

CONFIG_NEWLIB_LIBC is enabled in prj_newlibnano.conf, so the globals should be properly copied, from CMakeLists.txt:

```
  if(CONFIG_NEWLIB_LIBC)
    set(NEWLIB_PART -l libc.a z_libc_partition)
  endif()
```

Unless the library is not named libc.a?

ioannisg, 2019-11-14:
> CONFIG_NEWLIB_LIBC is enabled in prj_newlibnano.conf, so the globals should be properly copied, from CMakeLists.txt:
>
> ```
>   if(CONFIG_NEWLIB_LIBC)
>     set(NEWLIB_PART -l libc.a z_libc_partition) [...]

Right, it is called libc_nano.a.
:)

ioannisg, 2019-11-14:
Considering that we have a misconception of how the libc library is called, I'd raise this to Medium priority issue.

ioannisg, 2019-11-15:
@hakehuang can you tests the proposed fix in #20730 ?

hakehuang, 2019-11-16:
@ioannisg , #20730 works fine to me. log below

'''
***** Booting Zephyr OS build v2.1.0-rc1-131-g38e286aef050 *****

Running test suite test_c_lib_dynamic_memalloc

===================================================================

starting test - test_malloc

PASS - test_malloc

===================================================================

starting test - test_free

PASS - test_free

===================================================================

starting test - test_calloc

PASS - test_calloc

===================================================================

starting test - test_realloc

PASS - test_realloc

===================================================================

starting test - test_reallocarray

SKIP - test_reallocarray

===================================================================

starting test - test_memalloc_all

PASS - test_memalloc_all

===================================================================

starting test - test_memalloc_max

PASS - test_memalloc_max

===================================================================

Test suite test_c_lib_dynamic_memalloc succeeded

===================================================================

PROJECT EXECUTION SUCCESSFUL

'''

PR #20730 (2019-11-18): Test mem alloc fixes
Fixes #20598 

- minor change in CMakeLists.txt to copy globals from libc_nano.a
- tests/lib/mem_alloc updates for coverage increase
- minor typos and fixes
closed 2019-11-18 by MaureenHelm