zephyrproject-rtos/zephyr #20603: tests/kernel/critical failed on sam_e70_xplained board in v.1.14-branch

maksimmasalski, 2019-11-12:
Bug causing infinite loop and test can not be successfully finished.

Steps to reproduce the behavior:
1. Switch to the v.1.14-branch
2. $west build -p -b=sam_e70_xplained tests/kernel/critical
3. $west flash
4. See the error in minicom: $sudo minicom -D /dev/ttyACM0 -b 115200

***** Booting Zephyr OS zephyr-v1.14.1-24-gfbc0a44987a9 *****
Running test suite kernel_critical
===================================================================
starting test - test_critical
***** Booting Zephyr OS zephyr-v1.14.1-24-gfbc0a44987a9 *****
Running test suite kernel_critical
===================================================================
starting test - test_critical
***** Booting Zephyr OS zephyr-v1.14.1-24-gfbc0a44987a9 *****
Running test suite kernel_critical

Environment (please complete the following information):

    OS: Fedora28
    Toolchain: Zephyr-sdk-0.10.3
    Commit ID: fbc0a44987a94becf1dcaa6dc9b53a19857d0010

maksimmasalski, 2019-11-12:
Related issues #20445 
Some solutions discussed #20585

pabigot, 2019-11-12:
I can confirm this test passes with the solutions described in #20585 as long as

    -- -DCONFIG_WDT_DISABLE_AT_BOOT=y

is added to the west build command (and #20449 is fixed).  This option is applied to this target conditionally in `testcase.yaml`.

pabigot, 2019-11-12:
Actually, I also confirm this test passes on `v1.14-branch` at fbc0a44987a94becf1dcaa6dc9b53a19857d0010 without patches as long as the watchdog is disabled at build time.  So I don't believe there is a bug here.

@maksimmasalski  please re-check with the watchdog disabled.

maksimmasalski, 2019-11-12:
@pabigot 
Yes, I also did watchdog disable when tried to find out the problem, and it works well. Why master branch works without that configuration?

maksimmasalski, 2019-11-12:
@pabigot with disabled watchdog it works well.

maksimmasalski, 2019-11-12:
I can submit a PR to add watchdog disable into the configuration file for that board.

pabigot, 2019-11-12:
The test does not pass for me on master at 0297756342fb60e6a64813aac046f4d241cbd369 without  the patches described in #20585 *and* disabling the watchdog.  What master commit are you at?

I don't know whether it's appropriate to globally disable watchdog on that test given that it's in the testcase.yaml, though it makes sense to me so I'd approve it.  I've also pushed #20606 to provide meaningful diagnostics about test progress rather than 20 s of silence.

maksimmasalski, 2019-11-12:
I'm on the latest v.1.14-branch. You know, I have interesting behavior, if to change time in defines #define NUM_MILLISECONDS 3000 test passes successfully.

pabigot, 2019-11-12:
Weird, I'd have sworn I saw comments on here saying you could reproduce it on *master*.  I don't see those now.  It was 1am, but I thought I was awake.....

The SAM E70 default watchdog duration is 16 s, and with a 3 s loop the test completes before the watchdog fires.  Apparently the watchdog is stopped when the application completes, so the test doesn't loop restarting itself..
```
***** Booting Zephyr OS zephyr-v1.14.1-25-g842d8738cd9d *****
Running test suite kernel_critical
===================================================================
starting test - test_critical
Start reg1 at 10
End reg1 at 3010
Start alt1 at 3010
End alt1 at 6010
Enable timeslicing at 6010
Start reg2 at 6010
End reg2 at 9010
Start alt2 at 9010
End alt2 at 12010
PASS - test_critical
===================================================================
Test suite kernel_critical succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL
```

With the default 5 s loop the watchdog resets the board part way through the alt2 iteration, causing a loop:
```
***** Booting Zephyr OS zephyr-v1.14.1-25-g842d8738cd9d *****
Running test suite kernel_critical
===================================================================
starting test - test_critical
Start reg1 at 10
End reg1 at 5020
Start alt1 at 5020
End alt1 at 10020
Enable timeslicing at 10020
Start reg2 at 10020
End reg2 at 15020
Start alt2 at 15020
***** Booting Zephyr OS zephyr-v1.14.1-25-g842d8738cd9d *****
Running test suite kernel_critical
```

In summary there is no bug, just a missing configuration which is only present when running under sanitytest.  So I'm going to close this, and we can discuss in #20607 the best way to not waste people's time on a test that doesn't pass with the default configuration.

closed 2019-11-12 by pabigot