zephyrproject-rtos/zephyr #20604: log will be discarded before logging_thread scheduled once

morganzha, 2019-11-12:
**Is your enhancement proposal related to a problem? Please describe.**
A clear and concise description of what the problem is.

if enable LOG subsystem with  CONFIG_LOG_PROCESS_THREAD=y,
based on current design, log will be discard before logging_thread scheduled once, 
since logging thread is assigned lowest priority, there need to explicit sleep in main thread before using log APIs.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
please check if this can be improve to avoid this scenarios.
one suggestion is, put log thread as high priority thread when define and set back to lowest priority after init

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

wentongwu, 2019-11-13:
@morganzha can you please try this PR on your platform? Thanks

nordic-krch, 2019-11-21:
>  log will be discard before logging_thread scheduled

This is actually not true. Logs are queued from the start of the system (https://github.com/zephyrproject-rtos/zephyr/blob/c9e96b3c1976a0d938cc69d860789434a5ae3299/kernel/init.c#L504)

Logs cannot be processed until log thread is started and some may be dropped if buffer is too small and it overflows. Is that what you see or something else?

wentongwu, 2020-03-24:
I think we should close this.

closed 2020-03-24 by wentongwu