zephyrproject-rtos/zephyr #20618: Can unicast address be relayed when send message over gatt proxy?

david-foo, 2019-11-12:
I use bluez meshctl tool to connect the mesh net work which has 2 nodes(nodeA and nodeB). All provision and configuration process has been done well.
It seem meshctl sellect a node to connect to randomly. When it connect to nodeA it can send message with unitcast destination address to nodeA but can not send message with unicast destination address to nodeB. On the contrary side, when it connect to nodeB it can only send message with unicast address to nodeB. It seems message with unicast destination address can not be relayed by a node that do not has this element of address. The question is, is this the restrain by mesh protocol?
Also I found the group address can be relayed no matter what node you connet to.

aescolar, 2019-11-12:
CC @trond-snekvik @jhedberg

jhedberg, 2019-11-12:
@david-foo it sounds like the GATT Proxy feature is not enabled for these nodes. You need to enable it not just in Kconfig but also in the configuration server. You can either enable it by default in your bt_mesh_cfg_srv struct definition with `.gatt_proxy = BT_MESH_GATT_PROXY_ENABLED` or enable it after provisioning using a configuration client.

As long as the GATT Proxy is enabled in the build but not in the configuration server the nodes will still advertise using node identity (as opposed to network identity) which gives you access to the Proxy but no other nodes in the network. What you describe sounds like this. If it's something else then it may be a bug somewhere.

david-foo, 2019-11-13:
@jhedberg  I can make sure bt_mesh_cfg_srv.gatt_proxy has been enabled because bluz meshctl tool use gatt to do provision and proxy data transfer. If gatt_proxy is not enabled in build, bluez meshctl can not see the device at all.  
Let me list the process and commands I used in meshctl:

**nodeA**

discover_unprovisioned on
provision ddd30000000000000000000000000000

menu config
target 0100
appkey-add 1
bind 0 1 1003
pub_set 0100 ceef 1 0 0 1003

**nodeB**
restart meshctl if needed...

discover_unprovisioned on
provision ddd20000000000000000000000000000

menu config
target 0101
appkey-add 1
bind 0 1 1002
sub_add 0101 ceef 1002

Up to this step, meshctl, nodeA and nodeB have all stored the net and app info persistently. NodeA has a genneric level client and NodeB has a genneric level server, both of them have group address 0xceef. As we can see client mode has unicast address 0x0100 and server has unicast address 0x0101.

restart meshctl, then run connect 0 to connect to the mesh network. Repeat this process for times. Sometimes it shows nodeA is sellect but other times nodeB is sellect, this seem be randomly.

I send packet from meshctl to the two nodes. Take genneric level get message for example. Pseudo code like this:

If nodeA connected
	send(0x0101,OP_GENERIC_LEVEL_GET)
		nodeB do not respond OP_GENERIC_LEVEL_STATUS
	send(0xceef,OP_GENERIC_LEVEL_GET)
		nodeB do respond OP_GENERIC_LEVEL_STATUS

If nodeB connected
	send(0x0101,OP_GENERIC_LEVEL_GET)
		nodeB do respond OP_GENERIC_LEVEL_STATUS
	send(0xceef,OP_GENERIC_LEVEL_GET)
		nodeB do respond OP_GENERIC_LEVEL_STATUS

carlescufi, 2019-12-05:
Closing since it's an answered question.

closed 2019-12-05 by carlescufi