zephyrproject-rtos/zephyr #20621: Invalid baudrate on stm32 usart

maxvankessel, 2019-11-12:
**Problem**
Receiving invalid characters on host from STM32 USART transmitter

**Cause**
Invalid baudrate value is written in BRR register.

**Config**
I've programmed a F4 to use USART1 & 6 (on APB2).
APB2 runs on core clock, which is 180MHz.
PLL runs on HSE 20MHz

 **Steps to reproduce** 
Set core clock to 180MHz
Set APB2 to div 1
Enable USART1 (115200,8n1)
Boot (sample hello world for example)

_Writing `BRR = ClockRate / Baudrate` works to (temporary) fix this_

FRASTM, 2019-11-13:
Hello, can you please detail what you add to enable and configure UARTs on this board

maxvankessel, 2019-11-13:
At home I have a F407, at which it's not reproducible without overclocking it to 180MHz.
Never the less, it did try it.

```
$ cd zephyr/samples/hello_world
$ cmake -B ../hello_world_build -DBOARD=stm32f4_disco -DCONFIG_CLOCK_STM32_APB2_PRESCALER=1 -DCONFIG_UART_1=y -DCONFIG_CLOCK_STM32_PLL_N_MULTIPLIER=360 -GNinja
$ ninja -C ../hello_world_build flash
```
If `-DCONFIG_CLOCK_STM32_APB2_PRESCALER=2` the sample works fine

And the zephyr console need to be set to `UART_1` not sure if this can be done by cmdline

Changing `CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC=168000000` to 180MHz didn't made a difference

At work I use F446 which runs at 180MHz

Hope this helps

maxvankessel, 2019-11-13:
I found the line where it fails:
```#define __LL_USART_DIV_SAMPLING16_100(__PERIPHCLK__, __BAUDRATE__)     (((__PERIPHCLK__)*25)/(4*(__BAUDRATE__)))```

This is in `stm32f4xx_ll_usart.h` line 376
Calculation overflows `(__PERIPHCLK__)*25)`

FRASTM, 2019-11-13:
for F407, with HSE @ 8MHz as a source for the PLL, the VCO (= HSE / M * N = 2 * N) cannot exceed 432. With N multiplier set to 336, VCO = 672  (With N multiplier set to 360, VCO = 720)
On the other hand,  APB2 peripheral clock is limited to 84MHz :
- if CLOCK_STM32_APB2_PRESCALER is 1 the APB2 peripheral clock becomes 168MHz
- if CLOCK_STM32_APB2_PRESCALER is 2 the APB2 peripheral clock becomes 84MHz 

A combination to test could be :
M divider =4 (unchanged)
N multiplier = 168  (or 180 with overclocking)
P divider = 2
sysclock becomes 168MHz  (or 180 MHz with overclocking)
APB2 Prescaler = 2

F446 has another combination when PLL is sourced by the HSE @8MHz,
sysclock = 180 MHz(divider M = 8, multiplier N = 360, divider P = 2) 
and APB2 prescaler  = 2  for 90MHz as a peripheral clock

maxvankessel, 2019-11-13:
@FRASTM , I see, this has totally slipped my mind
> On the other hand, APB2 peripheral clock is limited to 84MHz

The same goes for the F446, max APB2 for 90MHz. 
It's physically impossible, so I understand my failure.

I'm a bit ashamed that I didn't noticed this. I'm very sorry to have wasted your time.

closed 2019-11-13 by maxvankessel