zephyrproject-rtos/zephyr #20632: call to bt_gatt_hids_init influences execution time of work queue

ppryga, 2019-11-12:
**Describe the bug**
When init of BLE stack and call to bt_gatt_hids_init influences of a execution of system work queue. 
It is related with db_hash_process that is submitted to work queue. Because it uses settings settings_save_one function to store entry. That call takes aroun 50ms.

**Expected behavior**
Do not block system work queue for such long time.

**Impact**
It may cause blockage of system work queue for mentioned time. It may cause problem with high frequency works that use work queue also.

ppryga, 2019-11-12:
I've added additional information about the issue in update section of description.

Vudentz, 2019-12-13:
I don't really get this one, the stack can only really tell if the services have changed by generating a hash and then proceed to compare to the stored one. As for calling db_hash_store doesn't setting_save_one actually checks if the value is the same and skip any IO, or that is not always the case, in which case we should probably compare to stored_hash before commiting anything.

joerchan, 2019-12-13:
> I don't really get this one, the stack can only really tell if the services have changed by generating a hash and then proceed to compare to the stored one. As for calling db_hash_store doesn't [...]

@Vudentz We had the same discussion here in a different issue and the conclusion there was that settings_save would not overwrite if the value was the same.
https://github.com/zephyrproject-rtos/zephyr/issues/20629#issuecomment-562073390

mfiumara, 2020-02-11:
This might be related to #22730.

joerchan, 2020-02-12:
> This might be related to #22730.

That's probably correct. The hash will have to calculated, but should not be stored if not changed.

ppryga, 2020-05-06:
@pdunaj  or @MarekPieta can you verify if the issue sill exist?

mfiumara, 2020-05-06:
@ppryga this is probably resolved in #22730. Could you retest with lastest master and see if the issue is still there?

joerchan, 2020-05-07:
Close this as I was told by @MarekPieta that the issue has been fixed.

closed 2020-05-07 by joerchan