zephyrproject-rtos/zephyr #20639: x_nucleo_iks01a3 sample is not working anymore after #20560 has been merged

avisconti, 2019-11-12:
x_nucleo_iks01a3 sample does not print any message out on the console. Seems stuck somewhere. This happened after #20560. It seems to me the #20560 is not the source of the problem, but simply highlighted an issue.

**To Reproduce**
I used a nucleo F401re board with a x_nucleo_iks01a3 shield.
Then:

cd samples/shields/x_nucleo_iks01a3/standard/
mkdir build && cd build
cmake -DBOARD=nucleo_f401re ..
make

No messages on console.
System is stuck.

I debugged a bit and it seems to me that, maybe, a  "k_sem_give(&data->device_sync_sem);" is missing in i2c v1 implementation.

avisconti, 2019-11-14:
Hello @FRASTM can you take a look to this issue?
To recreate it you would need a x_nucleo_iks01a3.

FRASTM, 2019-11-15:
As I did not find any X-NUCLEO-IKS01A3 here, I tried with a X-NUCLEO-IKS01A1 shield  and get the boot msg:

> X-NUCLEO-IKS01A1 sensor dashboard
> 
> HTS221: Temperature: 23.8 C
> HTS221: Relative Humidity: 41.5%
> LPS25HB: Pressure:99.0 kpa
> LIS3MDL: Magnetic field (gauss): x: 0.4, y: -0.3, z: 0.2
> LSM6DS0: Acceleration (m.s-2): x: -0.1, y: 0.1, z: 9.9
> 30:: lis3mdl trig 0

Is the pb you report specific to a x_nucleo_iks01a3 ?

avisconti, 2019-11-18:
> As I did not find any X-NUCLEO-IKS01A3 here, I tried with a X-NUCLEO-IKS01A1 shield and get the boot msg:
>
>
> Is the pb you report specific to a x_nucleo_iks01a3 ?

Yes.
I think the issue is related to the number of sensor requests to the I2C bus. 
x_nucleo_iks01a1 has too few sensors

avisconti, 2019-11-18:
@FRASTM 
Is it possible to get one x_nucleo_iks01a3?

FRASTM, 2019-11-19:
Yes, one shield x_nucleo_iks01a3 has been purchased and should arrive within the week

pabigot, 2019-11-19:
FWIW I can confirm this is specific to ST, as the demo and i2c scanner work with the IKA01A3 on Nordic.

avisconti, 2019-11-19:
> FWIW I can confirm this is specific to ST, as the demo and i2c scanner work with the IKA01A3 on Nordic.

@pabigot 
Which hash id are you using?

pabigot, 2019-11-19:
It works on current master at 667886d63ddb1a75e5c34b4dd407093b4dfc254d

avisconti, 2019-11-20:
> It works on current master at [667886d](https://github.com/zephyrproject-rtos/zephyr/commit/667886d63ddb1a75e5c34b4dd407093b4dfc254d)
> 
> ```
> west build -b nrf52_pca10040 -- -DSHIELD=x_nucleo_iks01a3
> ```

Can you quickly try with 732b815f ?
To be honest the issue disappeared also in my case with latest master even if there are no changes to i2c driver. It is a tricky issue.

pabigot, 2019-11-20:
Test command was:

    rm -rf build && west build -b nucleo_l476rg -- -DSHIELD=x_nucleo_iks01a3 && west flash

AFAICT the sample works, from just before 732b815f4be46a0e0c1b540e8274a38b345e7276 through current master 16ba258052552a65d4041b8b0bf1db02643b2b56, but only with a power-down reset.  A running example will fail if the reset button on the board is pressed.

The output suggests the LSM6DSO initialization doesn't work with soft restart.

I can reproduce the soft-reset failure on the nrf52_pca10040 as well, and there I get an I2C transfer error, so it's not an ST I2C problem but probably an issue with communicating with an LSM6DSO that had already been configured and started.

I think it'd be fair to close this issue, but whichever of us touches the lsm6dso next should confirm and fix the soft-reset problem.  It'll be on my task list as part of the GPIO API rework if nobody else gets to it first.

avisconti, 2019-11-21:
> Test command was:
> 
> ```
> rm -rf build && west build -b nucleo_l476rg -- -DSHIELD=x_nucleo_iks01a3 && west flash
> ```
> 
> AFAICT the sample works, from just before [732b815](https://github.com/zephyrproject-rtos/zephyr/commit/732b815f4be46a0e0c1b540e8274a38b345e7276) through current master [16ba258](https://github.com/zephyrproject-rtos/zephyr/commit/16ba258052552a65d4041b8b0bf1db02643b2b56), but only with a power-down reset. A running example will fail if the reset button on the board is pressed.
> 
> The output suggests the LSM6DSO initialization doesn't work with soft restart.
> 

No, this is known. The issue is related to the fact that INT1 pin is used to output the drdy interrupt, but after reset in input to enter in I3C hotjoin mode. So, since there is a level shifter on INT1 the '1' state is preserved when interrupt is generated. There are possible solutions like using either polling mode or INT2 for interrupt (instead of INT1).

> I can reproduce the soft-reset failure on the nrf52_pca10040 as well, and there I get an I2C transfer error, so it's not an ST I2C problem but probably an issue with communicating with an LSM6DSO [...]

OK, I will close this one. I can try to fix it right now with above considerations.

closed 2019-11-21 by avisconti

pabigot, 2019-11-21:
> No, this is known. The issue is related to the fact that INT1 pin is used to output the drdy interrupt, but after reset in input to enter in I3C hotjoin mode. So, since there is a level shifter on [...]

Oh, good.  Actually that's common, and the idiom I've been using in updating the sensors for the new GPIO API accommodates it.  Now that I have an A3 I can add that sensor to the set of example conversions.