zephyrproject-rtos/zephyr #20651: Bluetooth: disbable and restart BT functionality

codingspirit, 2019-11-13:
[This is a question]
Hi Zephyr community,

I'm trying to find a way to turn off/disable bt functionality(i.e. like `bt_disbale`) and restart it later on, but it looks like no such of API has been provided. Any hints will be appreciated. 

**To Reproduce**
Steps to reproduce the behavior:
1. Call `bt_enable`
2. Turn off hci controller
3. Power on hci controller
3. Trying to enable and use bt again
4. See error

In our case, we have a host and a controller, both of them are running zephyr with HCI H:4 and we want to cut off power of controller to save power sometime, then enable it again when needed. Then we found that we can't use any bt functionalities after re-powered HCI controller.

jhedberg, 2019-11-13:
@carlescufi FYI

jhedberg, 2019-11-13:
@codingspirit how exactly do you do steps 2 and 3? I suppose that's something hardware specific?

We do have plans to introduce a close() callback for HCI drivers, which would map to a new bt_disable() call, however so far it's just a plan, i.e. no code has been implemented yet. Would you be interested in trying to implement this?

jhedberg, 2019-11-13:
Btw, it looks like this is a duplicate of #3192 so we may want to close this one and keep the discussion in the original issue.

carlescufi, 2019-11-13:
Yes, this is a duplicate of #3192. @joerchan is currently working on this feature, so we can close this issue and track it in the original one. @codingspirit if you have additional input please add it there.

closed 2019-11-13 by carlescufi