zephyrproject-rtos/zephyr #20652: The z_arch_system_halt function should not dead loop by default

rakons, 2019-11-13:
**Is your enhancement proposal related to a problem? Please describe.**
When system halts without logger enabled it just silently dead loops, Used do not have any feedback.

**Describe the solution you'd like**
System should restart by default when no logger is enabled.

**Describe alternatives you've considered**
The z_arch_system_halt inside fatal.c file is weakly linked function, and the user may reimplement it (and probably should).
We can eventually overwrite this function in every example.
But the current default configuration is extremely not save for released code - it should be only used to development.

**Additional context**
If system restarts on system error, it makes it possible to recover. Going to dead loop is very bad behaviour as it would require to disconnect power from the device to recover. Additionally it drains the battery really fast. The current default implementation should never be used in released code.

nashif, 2019-11-13:
@andrewboie fyi

andrewboie, 2019-11-13:
We'll need to provide a arch_system_halt() implementation for all platforms, then we can simply delete the __weak version in kernel/fatal.c

rakons, 2019-11-14:
Can we at least force it to print anything if logging is disabled?

nashif, 2024-02-08:
closing, z_arch_system_halt no more in zephyr, if that got a new name, still needs to be addressed, etc, please reopen with new details.

closed 2024-02-08 by nashif