zephyrproject-rtos/zephyr #20654: Add documentation for the flash simulator

vanwinkeljan, 2019-11-13:
**Is your enhancement proposal related to a problem? Please describe.**

Currently there is no documentation about how to use the flash simulator

**Describe the solution you'd like**

Add documentation about how to use the flash simulator, eg. how to specify the flash size, sector size, features supported via KConfig, ...

**Describe alternatives you've considered**

Browse the code base

isaacjerish, 2025-05-01:
I'd like to work on this.

alexenr02, 2026-11-07:
This looks resolved by #120427 (merged 2026-10-29), which added a dedicated "Flash Simulator" section to doc/hardware/peripherals/flash.rst covering the devicetree node, its properties, the Kconfig options and usage. It is live under Hardware Support > Peripherals > Flash. Can this issue be closed?