zephyrproject-rtos/zephyr #20663: fdtable.c is being built always, regardless of usage

nashif, 2019-11-13:
Noticing that this file is building unconditionally, whether used or not. This takes 20 byes of RAM

[image]

nashif, 2019-11-13:
@pfalcon I remember you mentioning moving this somewhere else and making it conditional.

aescolar, 2019-11-14:
Almost duplicate of #12368

pfalcon, 2019-11-14:
As this is marked as priority: low, let me look into this when 2.2 window opens. The idea of solution would be to make building of fdtable.c  dependent of (hidden) Kconfig symbol, and `select` it where needed (currently POSIX, BSD Sockets).

andrewboie, 2019-11-14:
I think we should look at why gc-sections is being defeated here. If gc-sections was working as expected this would be a non-issue.

Notice that the only item pulled in is fdtable_lock. This is declared with K_MUTEX_DEFINE(). this ends up in a special section, defeating gc-sections, because K_MUTEX_DEFINE uses Z_STRUCT_SECTION_ITERABLE(k_mutex, name)

However, the *only* reason why k_mutex is routed to its own section is to support CONFIG_OBJECT_TRACING, which isn't even enabled. If we address this the problem goes away.

This issue is present for a lot of kernel object types, not just mutexes. k_sem, k_msgq, k_pipe, k_timer, k_stack, k_mailbox, k_queue are also affected, as they only need to be iterated at boot to support object tracing. grep for Z_STRUCT_SECTION_FOREACH() in kernel/ to see what I mean.

aescolar, 2019-11-15:
@pfalcon Even though @andrewboie 's PR fixes the RAM impact noted in this issue (and in other cases so it is welcome in itself), the fix you proposed would still be relevant for #12368 , and therefore also welcome

andrewboie, 2020-07-24:
This isn't specific to fdtable -- any kernel object instantiated and placed in a Z_ITERABLE_SECTION() can do this. In the case of fdtable it is its mutex. The definition uses KEEP() which causes the objects to never be discarded even if completely unused.

andrewboie, 2020-07-24:
I'll take a look at this for 2.4. https://github.com/zephyrproject-rtos/zephyr/pull/27117 is I believe another related problem.

carlescufi, 2020-07-27:
It's not only the RAM, it's also the ROM footprint that is increased by unconditionally executing this code. See my comment [here](https://github.com/zephyrproject-rtos/zephyr/pull/27117#issuecomment-664406994) for more details.

andrewboie, 2020-08-12:
I think the root cause of this is more or less here:

```
#define Z_LINK_ITERABLE(struct_type) \
        _CONCAT(_##struct_type, _list_start) = .; \
        KEEP(*(SORT_BY_NAME(._##struct_type##.static.*))); \
        _CONCAT(_##struct_type, _list_end) = .
```

For any use of `Z_ITERABLE_SECTION_RAM` or `Z_ITERABLE_SECTION_ROM`, `Z_LINK_ITERABLE` is used to match the input sections.

Unconditionally using `KEEP()` here is bad. We only need to use KEEP() in certain cases where the objects are by design not referenced directly by their symbol name in code. Things like initialization levels, device drivers, static threads, shell commands. Iterable sections which just contain ordinary kernel objects referenced directly in code should *not* be using `KEEP()` here because they will be kept around forever even if unused, pulling in other code/data as @carlescufi and other have observed.

The trivial fix:
```
diff --git a/include/linker/linker-defs.h b/include/linker/linker-defs.h
index 1bd9bd6249..5de7d62a6c 100644
--- a/include/linker/linker-defs.h
+++ b/include/linker/linker-defs.h
@@ -39,7 +39,7 @@
 #ifdef _LINKER
 #define Z_LINK_ITERABLE(struct_type) \
        _CONCAT(_##struct_type, _list_start) = .; \
-       KEEP(*(SORT_BY_NAME(._##struct_type##.static.*))); \
+       *(SORT_BY_NAME(._##struct_type##.static.*)); \
        _CONCAT(_##struct_type, _list_end) = .

 /* Define an output section which will set up an iterable area
```

Causes lots and lots of tests to fail. So I think we're going to need to introduce some new macros `Z_ITERABLE_SECTION_KEEP_RAM` `Z_ITERABLE_SECTION_KEEP_ROM` that preserves the implicit `KEEP()` and identify which of our existing iterable sections require it.

Related to this is a fair number of iterable sections in include/linker/ which are defined ad hoc not using `Z_ITERABLE_SECTION` macros, many of these have `KEEP()` in them as well and I suspect most of these are just copypasta and not because this was strictly required. These should be audited.

carlescufi, 2020-08-14:
@andrewboie I am confused. Isn't the root cause (at least of the ones I analyzed in [this pr](https://github.com/zephyrproject-rtos/zephyr/pull/27117#issuecomment-664406994) that we add a `SYS_INIT()` macro which the linker has no way of knowing it's not required, and that then pulls the code used in the `SYS_INIT` handler in?

andrewboie, 2020-08-14:
> @andrewboie I am confused. Isn't the root cause (at least of the ones I analyzed in [this pr](https://github.com/zephyrproject-rtos/zephyr/pull/27117#issuecomment-664406994) that we add a [...]

@carlescufi you are right something else is at work here, although it's not the `SYS_INIT()` all by itself that is the culprit, but some interaction with it.

For most of the build, we link with `--whole-archive`:

```
For each archive mentioned on the command line after the --whole-archive option, include every object file in the archive in the link, rather than searching the archive for the required object files. This is normally used to turn an archive file into a shared library, forcing every object to be included in the resulting shared library. This option may be used more than once.
```

for the core `kernel/`, we link that all the object files produced by that directory in a special way with `--no-whole-archive` which is supposed to cause unused code to be dropped even if it references data inside itself that has KEEP(). if there are no external references outside of `kernel/`, all of it is gone.

An example is `kernel/system_work_q.c`, it has a `SYS_INIT()` to fire up this workqueue, but if nothing outside of `kernel/` uses the system workqueue, the whole thing is dropped (or is supposed to be, haven't seen any bugs on that one at any rate)

Something is defeating this for `kernel/kheap.c`. the heap code itself is split between the kernel and lib/os/ so this may have something to do with it. I'm looking at this part some more today.

carlescufi, 2020-08-14:
> An example is `kernel/system_work_q.c`, it has a `SYS_INIT()` to fire up this workqueue, but if nothing outside of `kernel/` uses the system workqueue, the whole thing is dropped (or is supposed to be, haven't seen any bugs on that one at any rate)

Interesting, I did not know that.

>Something is defeating this for kernel/kheap.c. the heap code itself is split between the kernel and lib/os/ so this may have something to do with it. I'm looking at this part some more today.

And also `kernel/mempool_sys.c`.

PR #28486 (2020-09-19): linker: do not force keep common kernel objects
This uses the new macros so the common kernel objects can be
garbage collected if they are not referred directly anywhere
in the code. This is useful for reducing library data size
as not all library functions and their corresponding kobjects
are being used.

Fixes #20663
commit 4f106c32 (2020-09-19): linker: do not force keep common kernel objects

This uses the new macros so the common kernel objects can be
garbage collected if they are not referred directly anywhere
in the code. This is useful for reducing library data size
as not all library functions and their corresponding kobjects
are being used.

Fixes #20663
closed 2020-09-19 by nashif