zephyrproject-rtos/zephyr #20666: Unexpected UART Kconfig warnings during build

dbkinder, 2019-11-13:
**Describe the bug**
Did a clean build of samples/basic/blinky for the reel_board, and got a couple of unexpected warnings about UART configuration.

**To Reproduce**
Steps to reproduce the behavior:
```
~/zephyrproject/zephyr$ west build -p -b reel_board samples/basic/blinky
-- west build: build configuration:
       source directory: /home/dbkinder/zephyrproject/zephyr/samples/basic/blinky
       build directory: /home/dbkinder/zephyrproject/zephyr/build (created)
       BOARD: reel_board (origin: command line)
-- west build: generating a build system
Zephyr version: 2.0.99
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.8", minimum required is "3.4") 
-- Selected BOARD reel_board
-- Found west: /home/dbkinder/.local/bin/west (found suitable version "0.6.2", minimum required is "0.6.0")
-- Loading /home/dbkinder/zephyrproject/zephyr/boards/arm/reel_board/reel_board.dts as base
-- Overlaying /home/dbkinder/zephyrproject/zephyr/dts/common/common.dts
Device tree configuration written to /home/dbkinder/zephyrproject/zephyr/build/zephyr/include/generated/generated_dts_board.conf
Parsing Kconfig tree in /home/dbkinder/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/dbkinder/zephyrproject/zephyr/boards/arm/reel_board/reel_board_defconfig'
Merged configuration '/home/dbkinder/zephyrproject/zephyr/samples/basic/blinky/prj.conf'
Configuration saved to '/home/dbkinder/zephyrproject/zephyr/build/zephyr/.config'

warning: UART_CONSOLE (defined at drivers/console/Kconfig:46) was assigned the value 'y' but got the
value 'n'. You can check symbol information (including dependencies) in the 'menuconfig' interface
(see the Application Development Primer section of the manual), or in the Kconfig reference at
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_UART_CONSOLE.html (which is updated
regularly from the master branch). See the 'Setting configuration values' section of the Board
Porting Guide as well.

warning: the choice symbol UART_0_NRF_UART (defined at drivers/serial/Kconfig.nrfx:28) was selected
(set =y), but no symbol ended up as the choice selection.  You can check symbol information
(including dependencies) in the 'menuconfig' interface (see the Application Development Primer
section of the manual), or in the Kconfig reference at
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_UART_0_NRF_UART.html (which is updated
regularly from the master branch). See the 'Setting configuration values' section of the Board
Porting Guide as well.
-- Cache files will be written to: /home/dbkinder/.cache/zephyr
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/dbkinder/zephyr-sdk-0.10.3/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /home/dbkinder/zephyrproject/zephyr/build
-- west build: building application
[1/116] Preparing syscall dependency handling

[111/116] Linking C executable zephyr/zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       15036 B         1 MB      1.43%
            SRAM:        4004 B       256 KB      1.53%
        IDT_LIST:          72 B         2 KB      3.52%
[116/116] Linking C executable zephyr/zephyr.elf
~/zephyrproject/zephyr$ 
```

**Expected behavior**
A build of this simple sample with no warnings.

**Impact**
Annoyance and confusion

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04.3
 - Toolchain Zephyr SDK 0.10.3

dbkinder, 2019-11-14:
Similar unexpected warnings from a clean build of blinky for the sam_e7_xplained board:
```
~/zephyrproject/zephyr$ west build -p -b sam_e70_xplained samples/basic/blinky
-- west build: build configuration:
       source directory: /home/dbkinder/zephyrproject/zephyr/samples/basic/blinky
       build directory: /home/dbkinder/zephyrproject/zephyr/build (created)
       BOARD: sam_e70_xplained (origin: command line)
-- west build: generating a build system
Zephyr version: 2.1.0
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.8", minimum required is "3.4") 
-- Selected BOARD sam_e70_xplained
-- Found west: /home/dbkinder/.local/bin/west (found suitable version "0.6.1", minimum required is "0.6.0")
-- Loading /home/dbkinder/zephyrproject/zephyr/boards/arm/sam_e70_xplained/sam_e70_xplained.dts as base
Devicetree configuration written to /home/dbkinder/zephyrproject/zephyr/build/zephyr/include/generated/generated_dts_board.conf
Parsing Kconfig tree in /home/dbkinder/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/dbkinder/zephyrproject/zephyr/boards/arm/sam_e70_xplained/sam_e70_xplained_defconfig'
Merged configuration '/home/dbkinder/zephyrproject/zephyr/samples/basic/blinky/prj.conf'
Configuration saved to '/home/dbkinder/zephyrproject/zephyr/build/zephyr/.config'

warning: UART_CONSOLE (defined at drivers/console/Kconfig:47) was assigned the value 'y' but got the
value 'n'. You can check symbol information (including dependencies) in the 'menuconfig' interface
(see the Application Development Primer section of the manual), or in the Kconfig reference at
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_UART_CONSOLE.html (which is updated
regularly from the master branch). See the 'Setting configuration values' section of the Board
Porting Guide as well.

warning: USART_SAM (defined at drivers/serial/Kconfig.usart_sam:6) was assigned the value 'y' but
got the value 'n'. You can check symbol information (including dependencies) in the 'menuconfig'
interface (see the Application Development Primer section of the manual), or in the Kconfig
reference at http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_USART_SAM.html (which is
updated regularly from the master branch). See the 'Setting configuration values' section of the
Board Porting Guide as well.

warning: USART_SAM_PORT_1 (defined at drivers/serial/Kconfig.usart_sam:24) was assigned the value
'y' but got the value 'n'. You can check symbol information (including dependencies) in the
'menuconfig' interface (see the Application Development Primer section of the manual), or in the
Kconfig reference at
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_USART_SAM_PORT_1.html (which is
updated regularly from the master branch). See the 'Setting configuration values' section of the
Board Porting Guide as well.
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Cache files will be written to: /home/dbkinder/.cache/zephyr
-- Configuring done
-- Generating done
-- Build files have been written to: /home/dbkinder/zephyrproject/zephyr/build
-- west build: building application
[1/115] Preparing syscall dependency handling

[110/115] Linking C executable zephyr/zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:        8944 B         2 MB      0.43%
            SRAM:        3924 B       384 KB      1.00%
        IDT_LIST:         104 B         2 KB      5.08%
[115/115] Linking C executable zephyr/zephyr.elf
~/zephyrproject/zephyr$
```

dbkinder, 2019-11-14:
BTW, none of these UART warnings are given when building the hello_world sample, so it may just be some configuration settings for blinky that's exposing this warning.

ulfalizer, 2019-11-16:
Suspect what's going on is that `SERIAL_HAS_DRIVER` is `n`. [UART_CONSOLE](https://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_UART_CONSOLE.html) has it as a dependency.

CC @mnkp

mnkp, 2019-11-16:
I believe the problem is that 
```
$ cat samples/basic/blinky/prj.conf 
CONFIG_GPIO=y
CONFIG_SERIAL=n
```
`n` selects `CONFIG_SERIAL` whereas virtually all board configuration files `y` select the symbol and dependencies (like `UART_CONSOLE`) in their respective `_defconfig` file. The problem is not limited to the two boards mentioned in the issue. According to [Default board configuration](https://docs.zephyrproject.org/latest/guides/porting/board_porting.html#default-board-configuration) guidelines all major components on the evaluation board should be enabled so in this case the `samples/basic/blinky/prj.conf` seems to be in the wrong. If board configuration files used `Kconfig.defconfig` instead of `_defconfig` to configure uart/console it should be possible to turn it off but making this modification is not really worth the effort.

I believe we should remove `CONFIG_SERIAL=n` from `samples/basic/blinky/prj.conf`. The project will build serial driver even if it doesn't use one but that's in agreement with an aforementioned board configuration guidelines.

dbkinder, 2019-11-16:
Indeed removing the ``CONFIG_SERIAL=n`` from blinky/prj.conf eliminates the warning.  Thanks.  I'll submit a PR to do just that, but I suspect there are other samples that may have this same issue.

mnkp, 2019-11-16:
Doing a quick grep reveals that `samples/basic/minimal` is another project that does the same. Here things become more tricky since we actually want to build project with minimum size. To be able to do it in a clean way we would need to modify all board configuration files to rely on `Kconfig.defconfig` instead of `_defconfig` to configure uart/console. The expectations of `samples/basic/minimal` are not really in line with Default Board Configuration guidelines.

PR #20751 (2019-11-18): samples: remove CONFIG_SERIAL from blinky
As noted in the issue #20666 discussion, though the console is not used
by this sample app, it can't really be "disabled" using this Kconfig
option, so remove the CONFIG_SERIAL=n line.

Fixes: #20666
commit da963d94 (2019-11-18): samples: remove CONFIG_SERIAL from blinky

As noted in the issue #20666 discussion, though the console is not used
by this sample app, it can't really be "disabled" using this Kconfig
option, so remove the CONFIG_SERIAL=n line.

Fixes: #20666
closed 2019-11-18 by carlescufi