zephyrproject-rtos/zephyr #20669: Devicetree: allow one binding to match multiple compatibles

mbolivar, 2019-11-13:
**Is your enhancement proposal related to a problem? Please describe.**

I would like to be able to write a single devicetree bindings file that allows multiple compatibles to match it.

This will allow me to write a single binding for a single compatible family of devices made by multiple vendors. My particular use case is WS2812 and SK6812 LED controllers, so I want to be able to write:

```
compatible: ["worldsemi,ws2812-spi", "szledcolor,sk6812-spi"]
```

**Describe the solution you'd like**

The above should match devicetree nodes with either of the compatibles in the list.

**Describe alternatives you've considered**

I could write multiple bindings files that each have their own compatible and  include a common one. This feels inelegant and repetitive, however.

**Additional context**

N/A

mbolivar, 2019-11-13:
from @galak on slack, for the record:

> we need to spec if the list is tried as 'x' OR 'y' OR 'z' ... vs 'x' AND 'y' AND 'z' ...

ulfalizer, 2019-11-13:
I had a PR that added this earlier, but it was rejected then. Can't remember which PR it was part of ATM though.

ulfalizer, 2019-11-13:
> from @galak on slack, for the record:
> 
> > we need to spec if the list is tried as 'x' OR 'y' OR 'z' ... vs 'x' AND 'y' AND 'z' ...

Should be an OR thing I think. If you want to put other restrictions on `compatible`, `properties:` might be a better place to do it, since that's where restrictions on properties are normally added.

mbolivar, 2019-11-14:
> Should be an OR thing I think.

I think I agree about this. The variant-compatible scheme seems to be along those lines also.

pabigot, 2019-11-15:
Cross-link with #20289 and #19904.

ulfalizer, 2019-11-15:
I still need to find that old PR too. Had it all implemented, basically.