zephyrproject-rtos/zephyr #20671: ARC: remove scheduler code from arch layer

andyross, 2019-11-13:
The ARC implementation of SMP includes significant logic (see discussions in #20082) that on other architectures is part of the scheduler.  In principle the architecture code shouldn't be inspecting thread state like priorities, or making decisions about threads to switch to beyond those provided by the scheduler.

We should identify the requirements that are driving this and unify the implementations.  Right now things are fragile and likely to skew as the core kernel evolves.

andyross, 2019-11-13:
Most of the complaints are around the logic in z_arc_smp_switch_in_isr(), which seems to be duplicating what the scheduler has already done by checking a swap_ok flag, updating _current, and adjusting timeslicing.

That is all done by the scheduler for you.  What is supposed to happen is that you call z_arch_get_next_switch_handle() and just switch to the handle it presents.  If, like x86_64 does now, you don't have a "handle" based switch, you can store a thread struct pointer there.

andrewboie, 2020-08-20:
if left unattended this is going to cause problems, really need to resolve this for the next release.

ruuddw, 2020-08-20:
> if left unattended this is going to cause problems, really need to resolve this for the next release.

SMP needs work, e.g. I believe we see issues for 4-CPU systems that we do not see for 2-CPU configurations.
Now Wayne left, @evgeniy-paltsev will pick this up. But, please give us some time to handle the transition.
If there is any specific issue you have in mind (besides the tracing), please let us know.

evgeniy-paltsev, 2020-08-20:
Hmm, indeed ARC had some messy scheduler code living in ARC arch layer, but it is gone after 3f88ddd5499788272f4495e391a3c871feb5a821.

Now instead of scheduler stuff check in ASM code the **_z_get_next_switch_handle_** function is used (via __get_next_switch_handle_ wrapper).
Checks for _swap_ok_ and threads pointers manipulation (which were done in _**z_arc_smp_switch_in_isr**_ function from [arch/arc/core/arc_smp.c](https://github.com/zephyrproject-rtos/zephyr/blob/master/arch/arc/core/arc_smp.c)) are also dropped.

So, I'm not sure if this issue is valid anymore. @andyross, could you please comment it?

andrewboie, 2020-08-20:
@andyross is on vacation this week but I believe he's back next week.

andyross, 2020-08-27:
Yes, that PR did resolve most of this (and honestly I'd mostly forgotten about this particular bug).  I don't see why this can't be closed.

closed 2020-08-27 by ruuddw