zephyrproject-rtos/zephyr #20673: guiconfig not working properly?

novalisek, 2019-11-13:
**Describe the bug**
Hello,
I am trying to run` west build -t guiconfig`, make some changes, they are writen to zephyr/.config, but still not used by west build.

**To Reproduce**
```
west init test2
cd test2`
west update
cd zephyr
zephyr-env.cmd
west build -b nrf52810_pca10040 samples\bluetooth\peripheral
 builds fine
west build -t guiconfig
 select Boot options - MCUBoot support (BOOTLOADER_MCUBOOT)
 save
settins are saved to build\zephyr\.config
west build -b nrf52810_pca10040 samples\bluetooth\peripheral
 still no MCUBOOT support, same parition scheme
I have to manualy modify samples\bluetooth\peripheral\prj.conf and add
CONFIG_BOOTLOADER_MCUBOOT=y

```
then 
west build -b nrf52810_pca10040 samples\bluetooth\peripheral  starts to build with new flash partition schema  (and doesn't finish linking, because for nrf52810 FW partition size if 54kB, but compiled binary takes 140kB - does anybody know how to make it smaller? to remove debug info or something? but this is for anothe thread may be)

**Expected behavior**
changes I make in GUI should be used while building

**Impact**
I have to search config defines and do the editing of prj.conf manuly

**Environment (please complete the following information):**
 - OS: Windows 7 64bit
 - Toolchain Zephyr, Arm toolchain 9-2019-q4
 - Commit SHA or Version used:  aae2c84619585143841c0ceb2552423e7dfbb6a3

ulfalizer, 2019-11-13:
If `zephyr\.config` looks correct, then the problem is in CMake or west. `guiconfig` only deals with writing the configuration file.

I'm not sure what west expects re. manually modifying `zephyr\.config` (changing the configuration in the configuration interfaces works like manually modifying the file, except it's safer).

ulfalizer, 2019-11-13:
Could try leaving of the `-b <board>` in the second command maybe. Maybe it re-initializes things. I'm not a west expert though.

mbolivar, 2019-11-14:
> Could try leaving of the `-b <board>` in the second command maybe. Maybe it re-initializes things. I'm not a west expert though.

That's fine and won't recreate .config. No changes are needed in the west command line.

>  still no MCUBOOT support, same parition scheme

What do you mean by this? How can you tell there's no mcuboot support and the partition scheme is not changed?

I can't reproduce this at least on Linux. When I look at build/zephyr/nrf52810_pca10040.dts.pre.tmp after running the above steps, I see this:

```
$ grep -B 9 -A 1 code-partition build/zephyr/nrf52810_pca10040.dts_compiled 

	chosen {
		zephyr,console = "/soc/uart@40002000";
		zephyr,shell-uart = "/soc/uart@40002000";
		zephyr,uart-mcumgr = "/soc/uart@40002000";
		zephyr,bt-mon-uart = "/soc/uart@40002000";
		zephyr,bt-c2h-uart = "/soc/uart@40002000";
		zephyr,sram = "/soc/memory@20000000";
		zephyr,flash = "/soc/flash-controller@4001e000/flash@0";
		zephyr,code-partition = "/soc/flash-controller@4001e000/flash@0/partitions/partition@c000";
	};
$ grep -B 8 -A 1 code-partition build/zephyr/nrf52810_pca10040.dts.pre.tmp 
 chosen {
  zephyr,console = &uart0;
  zephyr,shell-uart = &uart0;
  zephyr,uart-mcumgr = &uart0;
  zephyr,bt-mon-uart = &uart0;
  zephyr,bt-c2h-uart = &uart0;
  zephyr,sram = &sram0;
  zephyr,flash = &flash0;
  zephyr,code-partition = &slot0_partition;
 };
```

Both of these indicate that the code was relinked into the appropriate partition for MCUboot.

@novalisek can you clarify what you mean by " still no MCUBOOT support, same parition scheme"? What exactly is going wrong?

novalisek, 2019-11-14:
@ulfalizer - changes are not reset, because if you run guiconfig again, they are still there and yes, may be I should have chosen another subject than  "guiconfig not working", it seems that it is working, but it's changes aren't used by west build

@mbolivar 
When you don't yous MCUBOOT, build ends like this:

```

Memory region         Used Size  Region Size  %age Used
           FLASH:      140130 B       192 KB     71.27%
            SRAM:       24396 B        24 KB     99.27%
        IDT_LIST:         136 B         2 KB      6.64%

region size = 192KB - thats the whole flash for nrf52810 

```
but when you use it, it ends like this:
```

Memory region         Used Size  Region Size  %age Used
           FLASH:      140642 B        52 KB    264.13%
            SRAM:       24396 B        24 KB     99.27%
        IDT_LIST:         136 B         2 KB      6.64c:/gnu_arm_embedded/bin/..
/lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: zephyr\zephyr
_prebuilt.elf section `_TEXT_SECTION_NAME_2' will not fit in region `FLASH'
c:/gnu_arm_embedded/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi
/bin/ld.exe: region `FLASH' overflowed by 87394 bytes
collect2.exe: error: ld returned 1 exit status
```
it is trying to fit into 52KB, that is correct to me, but the binary won't fit in, because its size is 138kB (but thats another problem - I don't know if it's not possible to use BLE + MCUBOOT with nrf52810 )

still DTS is same in both cases, both have  code-partition = &slot0_partition and both have 

```
slot0_partition: partition@c000 {
 label = "image-0";
 reg = < 0xc000 0xd000 >;
};
```
somehow it alters linker.cmd

```
diff --git "a/zephyr\\build\\zephyr\\linker.cmd.boot" "b/zephyr\\build\\zephyr\\linker.cmd.noboot"
index 6ae14c312d..02fe735eb2 100644
--- "a/zephyr\\build\\zephyr\\linker.cmd.boot"
+++ "b/test2\\zephyr\\build\\zephyr\\linker.cmd.noboot"
@@ -2,7 +2,7 @@
 _region_min_align = 32;
 MEMORY
     {
-    FLASH (rx) : ORIGIN = (0x0 + 0xc000), LENGTH = 0xd000
+    FLASH (rx) : ORIGIN = (0x0 + 0x0), LENGTH = (192*1K - 0x0)
     SRAM (wx) : ORIGIN = 0x20000000, LENGTH = (24 * 1K)
     IDT_LIST (wx) : ORIGIN = (0x20000000 + (24 * 1K)), LENGTH = 2K
     }
@@ -40,10 +40,10 @@ SECTIONS
  *(.iplt)
  }
    
- _image_rom_start = (0x0 + 0xc000);
+ _image_rom_start = (0x0 + 0x0);
     text :
  {
- . = 0x200;
+ . = 0x0;
  _vector_start = .;
  KEEP(*(.exc_vector_table))
  KEEP(*(".exc_vector_table.*"))

```

carlescufi, 2019-11-14:
Downgraded to question since it's unclear whether this is a bug.

novalisek, 2019-11-14:
@carlescufi I think it is a bug, may be not in guiconfig, but somewhere else and may be there should be another issue for that problem. I am new to Zephyr and don't know it into deep so I first thought it was guiconfig problem.  
BTW I see you are from Nordic Semi, can I ask if it is possible to use Zephyr+BLE+MCUBOOT on nrf52810, or is the flash too small? Or should I better ask at users@lists.zephyrproject.org or slack so that it's not too much off-topic for this issue?

mbolivar, 2019-11-14:
> still DTS is same in both cases, both have code-partition = &slot0_partition and both have

You're right. I see now that the way this works has changed since the last time I looked at the details on this. The chosen partition now takes effect via CONFIG_FLASH_LOAD_OFFSET and CONFIG_FLASH_LOAD_SIZE.

> I think it is a bug, may be not in guiconfig, but somewhere else and may be there should be another issue for that problem. I am new to Zephyr and don't know it into deep so I first thought it was guiconfig problem.

However, there's no bug here.

At least not as far as I can see at any layer (guiconfig, west, or the build system).

```
$ west build -b nrf52810_pca10040 -s samples/bluetooth/peripheral -p
$ grep FLASH_LOAD build/zephyr/.config
CONFIG_HAS_FLASH_LOAD_OFFSET=y
CONFIG_FLASH_LOAD_OFFSET=0
CONFIG_FLASH_LOAD_SIZE=0
```

versus

```
$ west build -b nrf52810_pca10040 -s samples/bluetooth/peripheral -- -DCONFIG_BOOTLOADER_MCUBOOT=y
$ grep FLASH_LOAD build/zephyr/.config
CONFIG_HAS_FLASH_LOAD_OFFSET=y
CONFIG_FLASH_LOAD_OFFSET=0xc000
CONFIG_FLASH_LOAD_SIZE=0xd000
```

> it is trying to fit into 52KB, that is correct to me, but the binary won't fit in, because its size is 138kB 

The build failure is to be expected since slot0 is too small to hold this sample. Its partition size is indeed 0xd000 (52 KiB), which is configured in nrf52810_pca10040.dts. So asking for MCUboot support from Kconfig reduces the amount of flash you have available for your application.

Please take a look at this video for a tutorial on how mcuboot works: https://www.youtube.com/watch?v=tv502SMhwYE

And see the zephyr devicetree documentation for how flash partitions are configured.

> (but thats another problem - I don't know if it's possible to use BLE + MCUBOOT with nrf52810, that would be a problem for me, if not, because thats the main feature I need - BLE + OTA )

Perhaps you need to configure out some other bluetooth features to make the application smaller, or change the partition map in nrf52810_pca10040.dts to allocate more space to slot0_partition and slot1_partition. (From the youtube video, they must be the same size).

I can't comment on what subset of bluetooth you can fit into 52 KiB (the default partition size). I don't know enough about bluetooth.

However, I can say that guiconfig is working, and so is the build system, so I'm going to close this issue.

Please feel free to ask for help from the community elsewhere if the existing documentation about partitions and configuring bluetooth is not enough to answer your questions.

novalisek, 2019-11-14:
@mbolivar I don't think everything is OK, because if you call guiconfig, select MCUBOOT support, then `CONFIG_BOOTLOADER_MCUBOOT=y` is written into .config, but 
```
CONFIG_FLASH_LOAD_OFFSET=0xc000
CONFIG_FLASH_LOAD_SIZE=0xd000
```
are not changed to proper values in the end ... mcuboot may be supported by firmware, but I think it will not work properly.

why does it work (not perfectly, because the binary is too big, but that was expected), when you add `CONFIG_BOOTLOADER_MCUBOOT=y` into prj.conf and doesn't work when you select it by guiconfig?  there must be something missing  ...   as you wrote, also  `west build -b nrf52810_pca10040 -s samples/bluetooth/peripheral -- -DCONFIG_BOOTLOADER_MCUBOOT=y`  is working and changes offset+size in .config,  after that, when you open guiconfig, you can also see those values in "Build and link features, linker options ..."  
So guiconfig is just saving and loading values it reads from .config, that seems to be OK and it is not a bug in guiconfig.

But then there should be something, that also corrects the other values, like it does when it is defined in prj.conf ...  Now I know that it is better to do all the things manualy by adding it to prj.conf or by west build cmake params, that will solve the dependecies (but then I still have to search for right config names in guiconfig)
But don't you think it would be nice if this was also done automaticaly after you exit guiconfig, or by calling some other command? Ofcourse you may have manualy entered values in .config which could be overwritten by this, but there could be some switch to accept it or do something else  ... 

May be it could be just enough to make some "guiconfig" for editing prj.conf, as I understand changes in prj.conf have highest priority ...

mbolivar, 2019-11-14:
OK, I finally see what you're saying.

So yes, the problem is somewhere between Kconfig and DT, not guiconfig itself. Very sorry for not understanding you.

@ulfalizer I think you might want to take a look at this again.

**steps to reproduce:**

1. build samples/bluetooth/peripheral or an nrf52 board (e.g. nrf52810_pca10040)
2. run guiconfig or menuconfig
3. set BOOTLOADER_MCUBOOT=y
4. save .config and exit

**expected behavior**

New .config after enabling BOOTLOADER_MCUBOOT in step 3 / step 4 should alter the linker configuration options so the load offset/size put the Zephyr image into the code partition.

E.g. for nrf52810_pca10040, you should see:

```
$ grep FLASH_LOAD build/zephyr/.config
CONFIG_HAS_FLASH_LOAD_OFFSET=y
CONFIG_FLASH_LOAD_OFFSET=0xc000
CONFIG_FLASH_LOAD_SIZE=0xd000
```

**actual behavior**

This is what I get instead:

```
$ grep FLASH_LOAD build/zephyr/.config
CONFIG_HAS_FLASH_LOAD_OFFSET=y
CONFIG_FLASH_LOAD_OFFSET=0
CONFIG_FLASH_LOAD_SIZE=0
```

So the FLASH_LOAD_OFFSET and FLASH_LOAD_SIZE values aren't getting reset properly after enabling BOOTLOADER_MCUBOOT.

ulfalizer, 2019-11-15:
If `FLASH_LOAD_OFFSET` and `FLASH_LOAD_SIZE` are user-assignable (have a prompt), then they will be "frozen" once `.config` is saved (e.g. by merging configurations) and then read back in.

The reason is that all Kconfig has to go on is this:

    CONFIG_FLASH_LOAD_OFFSET=0
    CONFIG_FLASH_LOAD_SIZE=0

If the symbol is user-configurable, then Kconfig has to assume that those were values that the user explicitly selected (e.g. by typing them into the configuration interface). If the symbols were not user-configurable, they would get recalculated from the defaults instead.

[Related PR](https://github.com/zephyrproject-rtos/zephyr/pull/20702) (and [related section on optional prompts](https://docs.zephyrproject.org/latest/guides/kconfig/index.html#optional-prompts)). Something similar should probably be done here.

This is a subtle thing has bitten people like five times now, so I should probably write it down on the Kconfig tips page so I can link to it. In general, always play around a bit in the `menuconfig` when adding new symbols. The `menuconfig` is the best representation of how Kconfig "thinks".

CC @SebastianBoe @joerchan

SebastianBoe, 2019-11-15:
I support applying the same fix to these options as we did in https://github.com/zephyrproject-rtos/zephyr/pull/20702 . But I'm not able to follow up on this.

carlescufi, 2019-11-15:
> @ulfalizer I've read your kconfig doc and I think the best solution for those two values would be to the last example, where it says hex "Stack size" if CUSTOM_STACK_SIZE) and make new symbols

Would you send a PR @novalisek ?

novalisek, 2019-11-15:
I can try to make one, but should it be like in #20702 or make new symbols like in @ulfalizer 's example ?

ulfalizer, 2019-11-15:
I could put one together. Need to look into these symbols first.

ulfalizer, 2019-11-16:
@novalisek @carlescufi @SebastianBoe 
PR: https://github.com/zephyrproject-rtos/zephyr/pull/20740

Also improved the documentation, as it seemed terse.

novalisek, 2019-11-16:
I tested #20740 with nrf52810_pca10040 - 

.config after new build with `west build -b nrf52810_pca10040 samples\bluetooth\peripheral`  without any changes via guiconfig:
```
CONFIG_TEXT_SECTION_OFFSET=0
# CONFIG_USE_DT_CODE_PARTITION is not set
CONFIG_FLASH_LOAD_OFFSET=0
CONFIG_FLASH_LOAD_SIZE=0
# CONFIG_BOOTLOADER_MCUBOOT is not set
```

after that, just switching MCUBOOT  ON in guiconfig sets
```
CONFIG_TEXT_SECTION_OFFSET=0x200
CONFIG_USE_DT_CODE_PARTITION=y
CONFIG_FLASH_LOAD_OFFSET=0xc000
CONFIG_FLASH_LOAD_SIZE=0xd000
CONFIG_BOOTLOADER_MCUBOOT=y
```

but after that, just switching MCUBOOT  OFF  in guiconfig sets
```
CONFIG_TEXT_SECTION_OFFSET=0x200
CONFIG_USE_DT_CODE_PARTITION=y
CONFIG_FLASH_LOAD_OFFSET=0xc000
CONFIG_FLASH_LOAD_SIZE=0xd000
# CONFIG_BOOTLOADER_MCUBOOT is not set
```
so you have mcuboot disabled, but the rest is wrong ...

ulfalizer, 2019-11-20:
@novalisek 
It's because Kconfig can't tell that

```
CONFIG_FLASH_LOAD_OFFSET=0xc000
CONFIG_FLASH_LOAD_SIZE=0xd000
```

isn't a user preference. Internally, it gets stored as the "user value" of the symbol, and once the symbol becomes visible (by toggling `MCUBOOT`), the user value is respected.

It's possible that the behavior could be different there (e.g. checking right after `.config` loading if a symbol is visible or not, and ignoring the user value if it isn't), but it would get pretty complex, and would break compatibility. The current behavior is compatible with how the C Kconfig tools work.

Note that an invalid configuration isn't generated in any of the cases from a Kconfig perspective. It's more that what's valid on the Kconfig end doesn't match what the code expects in this case.

ulfalizer, 2019-11-20:
"Visible" means that the symbol has a prompt (with a satisfied condition). It's called visible because it makes it visible in e.g. guiconfig/menuconfig.

ulfalizer, 2019-11-20:
The root cause of confusing stuff like this might be that Kconfig conflates the saved configuration with the configuration output. If the saved configuration used a different format, more information could be available.

I suspect splitting them up would still make Kconfig worse on the whole though. Most of the time it works great. Kconfig never generates an invalid configuration either, so a lot of it gets into gnarly details re. user expectations.

novalisek, 2019-11-20:
@ulfalizer 
and would it be possible to do it like I wrote in [#issuecomment-554242793](#issuecomment-554242793)

make new symbols  
```
CONFIG_CUSTOM_TEXT_SECTION_OFFSET
CONFIG_CUSTOM_FLASH_LOAD_OFFSET
CONFIG_CUSTOM_FLASH_LOAD_SIZE
```
and if they are not set, always use default values (that will depend on if MCUBOOT is enabled or not)?

Or is this not possible?

ulfalizer, 2019-11-20:
> @ulfalizer
> and would it be possible to do it like I wrote in [#issuecomment-554242793](#issuecomment-554242793)
>
> make new symbols
>
> ```
> CONFIG_CUSTOM_TEXT_SECTION_OFFSET [...]

Yeah, think it might be a better idea, if people would get bitten by `FLASH_LOAD_OFFSET` not reverting back to 0 when `MCUBOOT` is turned off.

What do you think about having a symbol like this instead?

```
config CUSTOM_FLASH_LOAD_OFFSET_SIZE
	bool "Use custom application load offset and size"
	help
	  If enabled, a custom kernel/application flash load offset and size
	  can be specified. When disabled, the load offset and size are
	  set automatically.
```

novalisek, 2019-11-20:
@ulfalizer 
Yes, I think it is not bad idea to have just one symbol for all 3 symbols  ... But don't also forget to unset USE_DT_CODE_PARTITION if MCUBOOT is not enabled

ulfalizer, 2019-11-20:
> @ulfalizer
> Yes, I think it is not bad idea to have just one symbol for all 3 symbols ... But don't also forget to unset USE_DT_CODE_PARTITION if MCUBOOT is not enabled

That might get trickier. There's no "anti-select" or the like (except `depends on` is kinda like an anti-select), and `USE_DT_CODE_PARTITION` is user-configurable. When Kconfig sees `CONFIG_USE_DT_CODE_PARTITION=y`, it has to assume that's what the user wanted, and can't turn it off when `BOOTLOADER_MCUBOOT` is turned off.

I wonder if the prompts on `FLASH_LOAD_OFFSET` and `FLASH_LOAD_SIZE` could be removed instead. Nothing sets them in a configuration file in-tree at least.

If they need to stay configurable, then having `CUSTOM_FLASH_LOAD_OFFSET_SIZE` as a separate option might be needed.

Maybe the existing fix could go in in the meantime. It's an improvement on the old situation at least.

ulfalizer, 2019-11-20:
No configuration file sets `USE_DT_CODE_PARTITION` either, fwiw...

ulfalizer, 2019-11-20:
Wonder if this might be a plan:

1. Get rid of the prompt on `USE_DT_CODE_PARTITION`, if there's no good reason to make it user-configurable

2. Add a new `CUSTOM_FLASH_LOAD_OFFSET_SIZE` symbol for when `FLASH_LOAD_OFFSET/SIZE` shouldn't be determined automatically, if needed

Alternatively, maybe the prompts on both `USE_DT_CODE_PARTITION` and `FLASH_LOAD_OFFSET/SIZE` could be removed, if there's no great reason to change them within the configuration interface in practice.

novalisek, 2019-11-20:
yes, `CONFIG_TEXT_SECTION_OFFSET`  and  `CONFIG_FLASH_LOAD_OFFSET` and `CONFIG_FLASH_LOAD_SIZE` prompts can be removed when `CUSTOM_FLASH_LOAD_OFFSET_SIZE` is not selected.  They don't have to depend on `USE_DT_CODE_PARTITION` , but few more files have dependencies on that (seeing in #20740 patch) .
It's a pitty there is no "unselect" property for symbols :) that would get called when unselecting that symbol. 
Or may be there could be values "a(auto),n(no),y(yes)" for boolean symbols instead just y,n and if it would see "a", than it would know it can be disabled because it was enabled automaticaly, not by user.
Then "select" property could also have one more parameter, so that it would look like
`select USE_DT_CODE_PARTITION a` - that would set it to "a(uto)"
and
`unselect USE_DT_CODE_PARTITION if a` - unselect if a(uto) 
then `!USE_DT_CODE_PARTITION` would also have to check if USE_DT_CODE_PARTITION is different from y(es) and a(uto)
but those are big changes :)  it's jus an idea :)

Now it will be easiest just to remove `USE_DT_CODE_PARTITION` prompt and add `CUSTOM_FLASH_LOAD_OFFSET_SIZE` symbol (I think someone may want to change the values), I would not remove them all.

ulfalizer, 2019-11-20:
@novalisek 
One thing that complicates things is that `.config` is also a configuration output format (it actually uses Makefile syntax, to allow configuration values to be read from `make`). If `CONFIG_FOO=a` or the like were to show up, then other tools reading the configuration would have to know how to deal with them.

I sometimes try to think of ways to make Kconfig a bit less confusing for people, but then I think of all the interactions with other things, and end up at "it's not so bad after all". :)

Often the root cause is some part of the configuration that could be improved as well, like things being configurable that don't need to be, and not an inherent issue with Kconfig.

novalisek, 2019-11-20:
@ulfalizer I don't know how make parses the file, may be this "auto" flag could be added after # character, that may be ignored by make, but it is probably also ignored by guiconfig etc, so it's not a good hack.
And what about making guiconfig for prj.conf? You could use more symbol options there (yes, no, unset, numerical value etc ..), because it is somehow parsed and merged with .config and this parser could be changed, may be ... then you wouldn't have to touch .config by guiconfig at all. 
But now I think it will be the best to remove `USE_DT_CODE_PARTITION` prompt and add `CUSTOM_FLASH_LOAD_OFFSET_SIZE` symbol

PR #20740 (2019-12-09): Kconfig: Fix FLASH_LOAD_{OFFSET,SIZE} issue when enabling BOOTLOADER_MCUBOOT in menuconfig + documentation and naming improvements
Fixes: #20673

Main fix:

```
kconfig: Make FLASH_LOAD_OFFSET/SIZE non-configurable when DT is used

Having FLASH_LOAD_OFFSET and FLASH_LOAD_SIZE always configurable froze
their values at 0 when BOOTLOADER_MCUBOOT was enabled in menuconfig,
when instead the values from /chosen/zephyr,code-partition in devicetree
should be used. BOOTLOADER_MCUBOOT selects USE_CODE_PARTITION, which is
a flag to use the devicetree information.

To fix it, only make FLASH_LOAD_OFFSET a [... cut]
commit 9cdb6f40 (2019-12-09): kconfig: Make FLASH_LOAD_OFFSET/SIZE non-configurable when DT is used

Having FLASH_LOAD_OFFSET and FLASH_LOAD_SIZE always configurable froze
their values at 0 when BOOTLOADER_MCUBOOT was enabled in menuconfig,
when instead the values from /chosen/zephyr,code-partition in devicetree
should be used. BOOTLOADER_MCUBOOT selects USE_CODE_PARTITION, which is
a flag to use the devicetree information.

To fix it, only make FLASH_LOAD_OFFSET and FLASH_LOAD_SIZE configurable
when USE_CODE_PARTITION is disabled. It looks like no configuration
files set them at the moment.

See the added documentation in [... cut]
closed 2019-12-09 by nashif