zephyrproject-rtos/zephyr #20686: tests/kernel/fp_sharing/float_disable failed when code coverage is enabled on qemu_x86.

chen-png, 2019-11-14:
**Describe the bug**
the common tests/kernel/fp_sharing/float_disable can pass on qemu_x86, bu if with the CONFIG_COVERAGE, it failed.

**To Reproduce**
Steps to reproduce the behavior:
1.  west build -b qemu_x86 tests/kernel/fp_sharing/float_disable/ -DCONF_FILE=prj_x86.conf 
     -DCOVERAGE=y
     or using sanitycheck -p qemu_x86 --coverage -T tests/kernel/fp_sharing/float_disable
2. west build -t run
3. See error

**Screenshots or console output**
SeaBIOS (version rel-1.12.1-0-ga5cab58e9a3f-prebuilt.qemu.org)
Booting from ROM..Optimal CONFIG_X86_MMU_PAGE_POOL_PAGES 11
E: General Protection Fault
E: Exception code: 0x0
E: EAX: 0x00160660, EBX: 0x00009500, ECX: 0x00000046, EDX: 0x00178578
E: ESI: 0x00000000, EDI: 0x00160b20, EBP: 0x00178540, ESP: 0x00178500
E: EFLAGS: 0x00000012 CS: 0x0008 CR3: 0x001602a0
E: call trace:
E: EIP: 0x00100338
E:      corrupted? (bp=0x00178540)
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x00178578 (unknown)
E: Halting system

**Environment (please complete the following information):**
 - OS: fedora28
 - Toolchain: zephyr-sdk-0.10.3
 - Commit ID: cb87cc8fd636

chen-png, 2019-11-20:
@andrewboie  after the "__aligned(0x20) struct x86_mmu_pdpt user_pdpt;" removed from arch/x86/include/ia32/kernel_arch_thread.h in #19968, this test case failed, when I added it, it will be ok.
do you know the root cause?

andrewboie, 2019-11-20:
> this test case failed, when I added it, it will be ok.
do you know the root cause?

That's very odd. It changes the size of the thread struct, but otherwise should have no effect.
What instruction is at 0x00100338 that caused the fault?

chen-png, 2019-11-20:
sorry, i don't know how to cat instruction according to address.
I add some gdb information below, is this useful?

0x00139871 in _handle_exc_13_vector_13_stub ()
Single stepping until exit from function _handle_exc_13_vector_13_stub,
which has no line number information.
_exception_enter () at   arch/x86/core/ia32/excstub.S:72

andrewboie, 2019-11-20:
several ways:
1. in gdb, "disassemble 0x00100338"
2. addr2line from shell.
3. look in build/zephyr/zephyr.lst

chen-png, 2019-11-20:
ok, thanks.
i ran the disassemble 0x00100338 in gdb, the information as follows:

Dump of assembler code for function arch_swap:
   0x00100320 <+0>:     push   %edi
   0x00100321 <+1>:     mov    $0x160b20,%edi
   0x00100326 <+6>:     push   %esi
   0x00100327 <+7>:     push   %ebx
   0x00100328 <+8>:     push   %ebp
   0x00100329 <+9>:     pushl  0x13fb60
   0x0010032f <+15>:    mov    0x8(%edi),%edx
   0x00100332 <+18>:    mov    %esp,0x28(%edx)
   0x00100335 <+21>:    mov    0x24(%edi),%eax
   0x00100338 <+24>:    fxsave 0x90(%edx)
   0x0010033f <+31>:    fninit
   0x00100341 <+33>:    fxrstor 0x90(%eax)
   0x00100348 <+40>:    mov    %eax,0x8(%edi)
   0x0010034b <+43>:    mov    0x28(%eax),%esp

andrewboie, 2019-11-22:
> 0x00100338 <+24>: fxsave 0x90(%edx)

This will GPF if the supplied memory address is unaligned. It's a bit mysterious how this is happening, but the solution for #17893 will fix this.

chen-png, 2019-12-04:
run "fxsave _thread_offset_to_preempFloatReg(%edx)",
(gdb) x $edx
0x178578 <_interrupt_stack+5496>:  0x00000000
and I find the _thread_offset_to_preemFloatReg is 0x90(0x80+0x10) in generated/offset.h file.
for the "fxsave m512byte" destination operand contains the first byte of the memory image, and it must be aligned on a 16-byte boundary. A misaligned destination operand will result in a general-protection (#GP) exception being generated.

chen-png, 2019-12-04:
without the CONFIG_COVERAGE,  I found the defined struct k_thread dummy_thread in z_cstart function address is 0x13e570<interrupt_stack+5488>, this is aligned to 16 bytes boundary.
(gdb) p _kernel
irq_stack=0x143800<interrupt_stack+26624>, current=0x13e570<interrupt_stack+5488>

however, when i enable the CONFIG_COVERAGE, the dummy thread address is 0x178578<interrupt_stack+5496>, not aligned to 16 bytes.
(gdb) p _kernel
irq_stack=0x17d800<interrupt_stack+26624>, current=0x178578<interrupt_stack+5496>

with the code #19968, it will influence the size and alignement of sttuct k_thread, so the dummy thread address is 0x178540<interrupt_stack+5440>, is also 16 bytes aligned.

jenmwms, 2020-09-23:
@andrewboie Did you prefer to keep this open for discussion? 

I suggest to close this, `tests/kernel/fpu_sharing/float_disable` is successful for qemu_x86 with and without coverage option enabled. Note the directory name changed slightly /fp_sharing/ -> /fpu_sharing/. 
```
$ west build -p  -b qemu_x86 tests/kernel/fpu_sharing/float_disable/ -DCONF_FILE=prj_x86.conf -DCONFIG_COVERAGE=y
...
$ west build -t run
-- west build: running target run
[0/1] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: qemu32,+nx,+pae
SeaBIOS (version rel-1.12.1-0-ga5cab58-dirty-20200516_152813-e7ee184eb100-zephyr
)
Booting from ROM..*** Booting Zephyr OS build zephyr-v2.3.0-2856-g1a7cd6ddbdcc  ***
Running test suite suite_k_float_disable
===================================================================
START - test_k_float_disable_common
 PASS - test_k_float_disable_common
===================================================================
START - test_k_float_disable_irq
Skipped for x86 or ARM without support for dynamic IRQs
 SKIP - test_k_float_disable_irq
===================================================================
START - test_k_float_disable_syscall
 PASS - test_k_float_disable_syscall
===================================================================
Test suite suite_k_float_disable succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL

andrewboie, 2020-09-23:
@jenmwms I think this is still an issue, we are just getting lucky. It all depends on the alignment of the dummy thread object that gets instantiated on the thread stack.

We need to somehow guarantee that the dummy thread on the call stack is 16-byte aligned.

It's not clear to me whether https://github.com/zephyrproject-rtos/zephyr/pull/19968 made this worse. But I haven't studied if the compiler makes any attempts to put aligned structs to their proper alignment when instantiated on the stack. The stack pointer isn't 16 byte aligned so I suspect not.

andrewboie, 2020-10-02:
never mind! I fixed this and forgot all about it:

```
/* The core kernel code puts the dummy thread on the stack, which unfortunately
 * doesn't work for 32-bit x86 as k_thread objects must be aligned due to the
 * buffer within them fed to fxsave/fxrstor.
 *
 * Use some sufficiently aligned bytes in the lower memory of the interrupt
 * stack instead, otherwise the logic is more or less the same.
 */
void arch_switch_to_main_thread(struct k_thread *main_thread, char *stack_ptr,
				k_thread_entry_t _main)
{
	struct k_thread *dummy_thread = (struct k_thread *)
		ROUND_UP(Z_KERNEL_STACK_BUFFER(z_interrupt_stacks[0]),
			 FP_REG_SET_ALIGN);

	__ASSERT(((uintptr_t)(&dummy_thread->arch.preempFloatReg) %
		  FP_REG_SET_ALIGN) == 0,
		 "unaligned dummy thread %p float member %p",
		 dummy_thread, &dummy_thread->arch.preempFloatReg);

	z_dummy_thread_init(dummy_thread);
	z_swap_unlocked();
	CODE_UNREACHABLE;
}
```

LixinGuoX, 2021-11-12:
We still have this problem when run the float_disable testcases with code coverage enabled.

closed 2022-01-26 by github-actions[bot]