zephyrproject-rtos/zephyr #20687: Clarification: How to enable on-board nor-flash following the board porting guide?

chrta, 2019-11-14:
This came up in #20194 and was raised by @mnkp in comment https://github.com/zephyrproject-rtos/zephyr/pull/20194#issuecomment-552673262 :

>  Since Default board configuration guidelines generally require to configure all major components available on the evaluation board maybe we should ensure that spi_nor driver is enabled if flash subsystem is enabled.

See https://docs.zephyrproject.org/latest/guides/porting/board_porting.html#default-board-configuration

On-board spi-nor-flash is present in some board's device tree, e.g. nrf52840_pca10056, see https://github.com/zephyrproject-rtos/zephyr/blob/master/boards/arm/nrf52840_pca10056/nrf52840_pca10056.dts#L194
But the configuration `CONFIG_SPI_NOR=y` is not set in the board's configuration, only in the sample drivers/spi_flash
This is the case for all boards with spi-nor-flash populated. It is only in the device tree, but the driver is not enabled in the configuration.

Is this indented that spi-nor-flash is not enabled in the board configuration?

If it is, i think it should be documented on the board_porting page.

pabigot, 2019-12-17:
I suspect it is intentional to leave the driver support out for boards that are primarily generic development platforms.  The SPI NOR devices are rarely used in these cases.

It's part of a wider issue in that these devices tend to have relatively high power draw (30% of Particle Xenon idle current goes to an unconfigured MX25L32 that isn't explicitly put into deep power-down mode).  #19448 needs attention to help solve that; it also suggests a need for drivers to support both `init` and `deinit` functions, and possibly only provide and invoke `deinit` based on devicetree status.

But yes, the "enable everything" directive should be reconsidered.    Sensors are another existing common exception.

chrta, 2020-09-20:
@pabigot I think this issue can be closed, since the board porting guide now contains this in https://docs.zephyrproject.org/latest/guides/porting/board_porting.html?highlight=board%20porting#general-recommendations

```
- Unless explicitly recommended otherwise by this section, leave peripherals and their drivers disabled by default.
```

pabigot, 2020-09-20:
OK.

closed 2020-09-20 by pabigot