zephyrproject-rtos/zephyr #20693: tests: watchdog: test_wdt_callback_1() implementation vs API specification

Oanerer, 2019-11-14:
**Describe the bug**
While trying to test my watchdog implementation for the Silicon Labs Gecko series [PR #20316 ](https://github.com/zephyrproject-rtos/zephyr/pull/20316) I noticed that the test implementation of ``test_wdt_callback_1()`` relies on a callback function being called by the watchdog hardware before the actual reset is performed when a watchdog timeout is reached. This implementation works for hardware such as NXP K64F or Nordic Semi NRF9160, which add a couple of watchdog clock cycles to execute callback functions before resetting. The Silicon Labs Gecko series do not have this feature and the reset takes precedence over the callback handling causing the watchdog to immediately reset and the test to fail.

**To Reproduce**
Steps to reproduce the behavior:
1. get [PR #20316 ](https://github.com/zephyrproject-rtos/zephyr/pull/20316)
2. west build -b efm32pg_stk3402a tests/drivers/watchdog/wdt_basic_api
3. west flash
4. See error

**Expected behavior**
Based on the API specification I would expect that when a callback function is set, the flag ``WDT_FLAG_RESET_NONE`` is also be passed to make sure the hardware does not reset, but only execute the callback function. This is not how the test is implemented though. In it's current form, when setting a callback function AND passing flag ``WDT_FLAG_RESET_SOC`` (as in ``test_wdt_callback_1()``) the outcome is inconsistent across different hardware implementations. 

**Impact**
Cannot succesfully complete watchdog test

**Screenshots or console output**
```Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
I: Installed timeout value: 2049
Waiting to restart MCU
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Testcase passed
Testcase: test_wdt_callback_1
I: Installed timeout value: 2049
Waiting to restart MCU
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_callback_1

    Assertion failed at ZEPHYR_BASE/tests/drivers/watchdog/wdt_basic_api/src/test_wdt.c:312: test_wdt: test_wdt_callback_1() == TC_PASS is false

FAIL - test_wdt
===================================================================
Test suite wdt_basic_test failed.
===================================================================
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: (ArchLinux using W10 WSL)
 - Toolchain: Zephyr SDK 0.10.3
 - [PR #20316 ](https://github.com/zephyrproject-rtos/zephyr/pull/20316)

**Additional context**
Additional information specific to the Gecko implementation can be found here: https://github.com/zephyrproject-rtos/zephyr/pull/20316#issuecomment-552812097

pabigot, 2019-11-19:
This was assigned medium in triage, but on inspection reproducing it requires the unmerged #20316 which adds watchdog support.  I'm dropping this to low so it doesn't impact the 2.1 release.

https://github.com/zephyrproject-rtos/zephyr/pull/20316/files#r348128121 may be sufficient to address this without changing the test (though the test may need to be adapted to  `-ENOTSUP` and bypass the test case rather than fail).  However, it may also be appropriate to extend the test to verify that the callback is invoked when no reset is performed.  It may also be necessary to assess whether the watchdog API provides the right access to the capabilities of additional watchdog implementations.

pabigot, 2020-06-29:
Reproduced on current master, fix in  #26521.

PR #26521 (2020-07-29): tests: wdt_basic_api: pass when requested test is not supported
Not all watchdog devices support a warning callback before reset.  If the driver refuses to accept that configuration record the test as having passed.

NOTE: This fix allows the second callback test to pass on these boards, but does not reset.  Thus a third callback test would not be invoked.  This should be OK because the third callback test would also be rejected.

Fixes #20693
closed 2020-07-29 by nashif