zephyrproject-rtos/zephyr #20697: Confusing warning during cmake

pfl, 2019-11-14:
While implementing new functionality using UARTs, I on purpose enabled the Kconfig variable CONFIG_UART_ASYNC_API. I turns out that this is a variable that has to have supporting code and other Kconfig options setting it, and is not meant to be enabled in prj.conf. The message telling me this is:
```
warning: SERIAL_SUPPORT_ASYNC (defined at drivers/serial/Kconfig:25)
was assigned the value 'y' but got the value 'n'. This symbol has no prompt,
meaning assignments in configuration files have no effect on it. It can only be
set indirectly, via Kconfig defaults (e.g. in a Kconfig.defconfig file) or through
being 'select'ed or 'imply'd (note: try to avoid Kconfig 'select's except for
trivial promptless "helper" symbols without dependencies, as it ignores
dependencies and forces symbols on).
You can check symbol information (including dependencies) in the
'menuconfig' interface (see the Application Development Primer section of
the manual), or in the Kconfig reference at
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_SERIAL_SU
PPORT_ASYNC.html (which is updated regularly from the master branch).
See the 'Setting configuration values' section of the Board Porting Guide as
well. It covers Kconfig.defconfig files.
```
The bug here is that this message is overly verbose and very confusing.

A message simply saying 'The Kconfig variable SERIAL_SUPPORT_ASYNC needs to have supporting code enabling it and cannot be set in a project prj.conf file' or similar would be enough and right on the spot. The cmake step causing this should definitely stop with an error and not pretend that everything went smoothly, especially since the message itself might have scrolled out of view in a developer's terminal. And it is decently grave by not producing the desired result, so it definitely cannot be classified only as a warning.

A similar issue is filed in https://github.com/zephyrproject-rtos/zephyr/issues/20666.

SebastianBoe, 2019-11-14:
It is not-a-bug that the warning text can be improved.

ulfalizer, 2019-11-14:
@pfl 
I wouldn't have understood what "supporting code" meant if I didn't already know Kconfig, and I think it's useful to explain what goes on on the Kconfig end, since you'd often have to understand Kconfig to debug it.

Maybe the "high-level" view could be explained better though, somewhere early on. I don't want to remove information though. In general, I think the worshipping of terseness in IT does way more harm than good.

ulfalizer, 2019-11-14:
The message also applies to other symbol types by the way (`string`/`int`/`hex`/`tristate` - though Zephyr doesn't use `tristate` symbols), so it might need to be tailored depending on the symbol type if a "high-level" explanation is added.

A symbol could also end up as `y` even though you set it to `n`, if it's being selected.

The reason the message is pretty generic now is to account for all those different cases. Doesn't mean it couldn't be tailored for specific cases though.

ulfalizer, 2019-11-14:
Oh, wait, I just remembered that it's already being tailored for the no-prompt case there, so it wouldn't be so bad in this case. I'll tinker a bit later.

The code that generates the warning is in `scripts/kconfig/kconfig.py`.

The reason it hasn't been turned into an error is that it'd be a lot of work to go through and fix everything. Many of those warnings are harmless too, but it could still be fixed.

I've spent lots of time cleaning up other things and turning them into errors. Maybe some day...

ulfalizer, 2019-11-14:
Think I figured out a nicer way to do things now, which could indirectly be used to improve the message.

The way the error is detected at the moment is just through the assigned value differing from the actual value, along with some slight special-casing of the error message for promptless symbols.

Better would be to check directly if a promptless symbol is assigned to, and always making that an error. Unfortunately, it gets complicated once a merged configuration has been written to `zephyr/.config` and gets read back in, because `zephyr/.config` includes values for promptless symbols too.

Instead, the code should do a separate check when `zephyr/.config` doesn't exist and gets generated by merging configuration files. In that case, it's safe to just check if some promptless symbol was assigned to, and turn that into an error. That'd also make the check more robust, because it would detect assignments to promptless symbols even when the actual value just happens to end up the same as the assigned value.

ulfalizer, 2019-11-15:
Found some stuff with the improved check: https://github.com/zephyrproject-rtos/zephyr/pull/20717 https://github.com/zephyrproject-rtos/zephyr/pull/20718 https://github.com/zephyrproject-rtos/zephyr/pull/20719 https://github.com/zephyrproject-rtos/zephyr/pull/20720

Will wait for those to be in, and then make a PR for the more robust check and tweak the message at the same time.

pfl, 2019-11-15:
Looks like a plan; if the value is promptless, it indeed could spit out an error. Let's see after those other Kconfig fixes. Thanks!

carlescufi, 2019-11-15:
@pfl should we close this now?

ulfalizer, 2019-11-16:
The change in https://github.com/zephyrproject-rtos/zephyr/pull/20742 improves the message.

pfl, 2019-11-18:
Yes, said change, when merged, will improve this message.

commit bb4a4e79 (2020-01-17): scripts: kconfig.py: Detect all assignments to promptless symbols

Assigning to promptless symbols has no effect.

Previously, the only check was for whether the value assigned to a
symbol matched its final value. This misses cases where a promptless
symbol is assigned to and just happens to get the assigned-to value as
its final value.

Instead, detect whether configuration files are being merged (by
checking if zephyr/.config already exists), and explicitly check for
assignments to promptless symbols in that case. We can't do it when
zephyr/.config already exists (and is being loaded), becau [... cut]
closed 2020-01-17 by carlescufi