zephyrproject-rtos/zephyr #20699: Each board should have a list of Kconfig options supported

pfl, 2019-11-14:
While working on an for me unknown board, let's say for example FRDM k64f, it would be excellent to have all the possible board specific Kconfig options listed. I was looking for UART support, and figured out from the header file that the asynchronous UART API would be the best fit for my project.

Unfortunately the only way of figuring out whether said example board supports this particular UART API is to dig through several orders of Kconfig files in order to learn whether the #ifdef in the include file is supported or not or whether to redesign ones project to use the more basic API.

In order to solve this, I propose a script crawling through the board specific Kconfig options and other included Kconfig files that would output decently formatted documentation added for each board. That way the connection between the APIs documented in include/ can be mapped to board specific support - or actually vice versa.

aescolar, 2019-11-14:
CC @ulfalizer

carlescufi, 2020-03-31:
This is already in `menuconfig` or `guiconfig`. A list would be extremely long and not hierarchical, so I am not sure this would be usable. Closing for now, if you have a concrete proposal please reopen it.

closed 2020-03-31 by carlescufi