zephyrproject-rtos/zephyr #20707: Define GATT service at run-time

kristoftjonck, 2019-11-14:
**Problem**
How am i able to create a custom GATT service with custom characteristics at run-time without predefining any memory space?

I based myself on the code found in hrs example from 1.14 to create a static implementation of the custom service & characteristics (see second code example). But when i create them on runtime (see added source code) i get an error. 

Note: now using the 2.0.99 Zephyr stack

**Not working code sample code what i'm trying to accomplish on runtime**
`CONFIG_BT_GATT_DYNAMIC_DB=y`

```
int bt_gatt_static_init(u8_t n)
{
	//Make custom ID according to number n
	struct bt_uuid_128 BT_UUID_CUSTOM =
	BT_UUID_INIT_128( n, 0x00, 0x00, 0x00, 0x00, 0x00 ,0x00, 0x00, 0x00
			,0x00 ,0x00 ,0x00 ,0xFF, 0xFF, 0xFF, 0xFF);
	
	//Create attributes and register service
	struct bt_gatt_attr custom_attrs[] = {
			BT_GATT_PRIMARY_SERVICE(&BT_UUID_CUSTOM),
			//Variable amount of characteristics
	};
	struct bt_gatt_service custom_svc = BT_GATT_SERVICE(custom_attrs);
	return bt_gatt_service_register(&custom_svc);
}
```

Running this code at runtime (triggered by reading an other characteristic) gives following error: 
```
[00:00:42.246,612] <err> os: ***** BUS FAULT *****                              
[00:00:42.246,643] <err> os:   Precise data bus error                           
[00:00:42.246,643] <err> os:   BFAR Address: 0xfe441956                         
[00:00:42.246,643] <err> os: r0/a1:  0x00000001  r1/a2:  0x00000002  r2/a3:  0xc
[00:00:42.246,673] <err> os: r3/a4:  0xfe441946 r12/ip:  0x00000016 r14/lr:  0xf
[00:00:42.246,673] <err> os:  xpsr:  0x21000000                                 
[00:00:42.246,673] <err> os: Faulting instruction address (r15/pc): 0x00009c6a  
[00:00:42.246,673] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception            
[00:00:42.246,673] <err> os: Current thread: 0x20001a20 (unknown)               
[00:00:42.307,464] <err> os: Halting system
```

**Static code**
```
static struct bt_uuid_128 BT_UUID_CUSTOM =
BT_UUID_INIT_128( 0x01, 0x00, 0x00, 0x00, 0x00, 0x00 ,0x00, 0x00, 0x00
        ,0x00 ,0x00 ,0x00 ,0xFF, 0xFF, 0xFF, 0xFF);
 
//Create attributes and register service
static struct bt_gatt_attr custom_attrs[] = {
        BT_GATT_PRIMARY_SERVICE(&BT_UUID_CUSTOM),
        //Variable amount of characteristics
};
static struct bt_gatt_service custom_svc = BT_GATT_SERVICE(custom_attrs);
int bt_gatt_custom_init(u8_t num) {
    return bt_gatt_service_register(&custom_svc);
}
```

joerchan, 2019-11-14:
I'm not sure exactly what you are trying to do. If you need dynamic storage for your service you will have to use `k_malloc`, because the example you gave declares the memory on the stack, which will leave to invalid memory, and eventually the bus fault you got.

If you want a dynamic service with static storage then you Static code example is correct for this use-case. This is not the static use case.
A static service with static storage would look like this:
```
BT_GATT_SERVICE_DEFINE(custm_svc,
        BT_GATT_PRIMARY_SERVICE(&BT_UUID_CUSTOM),
        //Variable amount of characteristics
```

kristoftjonck, 2019-11-19:
@joerchan I am indeed trying to get a dynamic service with dynamic storage. I tried using k_malloc() but still have some problems. I tried some possibilities but can't get it to work. On the android app the following code adds a random descriptor to the last service (https://imgur.com/3R5HMEU) instead of a new service. 

```
int bt_gatt_dynamic_init(u8_t n) {
	/*
	 * Allocate for service UUID and characteristics
	 */
	struct bt_uuid_128 * BT_UUID_CUSTOM = k_malloc(sizeof(struct bt_uuid_128));
	if ( BT_UUID_CUSTOM == NULL){
		printk("COULD NOT ALLOCATE UUID");
	}
	struct bt_uuid_128 BT_UUID_CUSTOM_TEMP =
		BT_UUID_INIT_128( n, 0x00, 0x00, 0x00, 0x00, 0x00 ,0x00, 0x00, 0x00
				,0x00 ,0x00 ,0x00 ,0xFF, 0xFF, 0xFF, 0xFF);

	memcpy (BT_UUID_CUSTOM, &BT_UUID_CUSTOM_TEMP, sizeof (struct bt_uuid_128));

	/*
	 * Allocate and create attribute table
	 */
	struct bt_gatt_attr local_custom_attrs[] = {
			BT_GATT_PRIMARY_SERVICE(&BT_UUID_CUSTOM),
			//TODO characteristics
	};
	struct bt_gatt_attr * custom_attrs  = k_malloc(ARRAY_SIZE(local_custom_attrs) * sizeof(struct bt_uuid_128));
	if ( custom_attrs == NULL){
		printk("COULD NOT ALLOCATE ATTRIBUTE TABLE");
	}
	memcpy(custom_attrs, local_custom_attrs, ARRAY_SIZE(local_custom_attrs) * sizeof(struct bt_gatt_attr));

	/*
	 * Allocate and create custom service
	 */
	struct bt_gatt_service * custom_svc = k_malloc(sizeof(struct bt_gatt_service));
	if (custom_attrs == NULL){
		printk("COULD NOT ALLOCATE SERVICE");
	}
	struct bt_gatt_service custom_svc_local;
	custom_svc_local.attrs = custom_attrs;
	custom_svc_local.attr_count = ARRAY_SIZE(local_custom_attrs);
	memcpy (custom_svc, &custom_svc_local, sizeof (struct bt_gatt_service));

	printk("Registering custom service");
	return bt_gatt_service_register(custom_svc);
}
```

joerchan, 2019-11-20:
@kristoftjonck You are on the right track. Your code still has some issues though.

```
struct bt_gatt_attr local_custom_attrs[] = {
			BT_GATT_PRIMARY_SERVICE(&BT_UUID_CUSTOM),
			//TODO characteristics
	};
	struct bt_gatt_attr * custom_attrs  = k_malloc(ARRAY_SIZE(local_custom_attrs) * sizeof(struct bt_uuid_128));
```
This malloc call has is using `sizeof(struct bt_uuid_128)` So the size in this case will be wrong.
Also there are other UUIDs currently on the stack masked by the use of BT_GATT_PRIMARY_SERVICE and probably BT_GATT_CHARACTERISTIC.

So just continue with what you are doing and make sure that you get everything that is currently declared on the stack into valid memory.

closed 2019-11-22 by joerchan

CodingGhost, 2021-05-13:
I would also need this, @kristoftjonck did you solve it?

joerchan, 2021-05-18:
@CodingGhost The `BT_GATT_PRIMARY_SERVICE` has declared it's UUID with `BT_UUID_GATT_PRIMARY` which is now on the stack.
So you would have to re-assign this pointer to something not on the stack.

```
static struct bt_uuid const * const uuid_primary = BT_UUID_GATT_PRIMARY;
...
	attr = k_malloc(sizeof(struct bt_gatt_attr));
	*attr = (struct bt_gatt_attr)BT_GATT_PRIMARY_SERVICE(uuid);
	attr->uuid = uuid_primary;
```

joerchan, 2022-10-10:
I suggest you ask this question on nordics support forum: [devzone.nordicsemi.no](https://devzone.nordicsemi.com/)
Or you can get in touch with the zephyr community: https://zephyrproject.org/community/#connect

Asking questions on closed issues is not where you would get answers.