zephyrproject-rtos/zephyr #20708: Blocking clock enable API for drivers/clock_control.h

mbolivar, 2019-11-14:
**Describe the bug**

We should have a synchronous clock enable API.

**Background**

When the clock_control.h API was added way back in the mists of time (November 2015, in 714dd9786b91a33374c79a01dcc39554130e810c), it had only two functions:

- `clock_control_on()`
- `clock_control_off()`

No mention was made in the API documentation about whether or not the clock enabled by `clock_control_on()` was ready by the time the call returned. There was no API for getting the status of a clock at this time, either.

More recently (June 2019, in e0943cfaab29100c5212f74aa9f3eb1a4aa58f0b), the API was extended with:

- `clock_control_async_on()`
- `clock_control_get_status()`

The `clock_control_async_on()` API is unambiguously asynchronous, but the `clock_control_on()` API documentation remains unclear.

In practice, it seems that most driver implementations for `clock_control_on()` are non-blocking (I checked Nordic's, and @carlescufi says others he checked were also).

**Expected behavior**

The non-async `clock_control_on()` API should clearly state whether it is blocking or not.

A blocking API should exist if it's non-blocking.

It's arguably wrong that `clock_control_on()` isn't blocking, since when it was introduced there was no way for the user to know when the clock was ready (since `clock_control_get_status()` was not added until almost 4 years later). The user thus had to either break the abstraction to query a lower level HAL API directly to find that out, or just hope that the clock would be enabled "soon enough" for their purpose.

With the async API, if a user wants to block until a clock is enabled, they have two choices:

1. Always use `clock_control_async_on()` and sleep until awoken by the callback
2. Block in a loop on `clock_control_get_status()`

But it's an annoyance for callers to have to reimplement this by themselves. And it's ambiguous and misleading that the non-asynchronous API isnt, ... well, synchronous.

**Expected behavior**

It would be nicer to have an explicitly blocking clock control enable API.  It would be better still if this API could let the caller choose whether to busy-wait or sleep until the clock was enabled.

**Impact**

Annoyance

**Environment (please complete the following information):**

Since e0943cfaab29100c5212f74aa9f3eb1a4aa58f0b.

tbursztyka, 2019-11-14:
> It's arguably wrong that clock_control_on() isn't blocking, since when it was introduced there was no way for the user to know when the clock was ready (since clock_control_get_status() was not added until almost 4 years later). 

When it was added it was indeed meant to be blocking. So when it returned 0, user knew the on/off worked. Documentation was missing (we were doing a lot of API without any docs at that time).

The common pattern (in every APIs) is that you have blocking functions *unless* having _async added to their names.

Now, if many drivers are non-blocking on these non-async functions: that's a bug.

nordic-krch, 2019-11-15:
I agree. It makes sense to make it blocking and consistent with other drivers.